google/skywater-pdk #172: Missing attributes in LEF files (.magic.lef) for VNB and VPB pins

agorararmard, 2020-10-08:
## Expected Behavior
The USE and DIRECTION sections should be added because this is causing parsing errors and warnings with some of the opensource tools, i.e.: OpenROAD and OpenPhySyn.

```
PIN VPB
    USE POWER ;
    DIRECTION INOUT ;
    PORT
      LAYER nwell ;
        RECT -0.190000 1.305000 2.950000 2.910000 ;
    END
  END VPB
```
For VNB it should be: `USE GROUND ;`

I'm not sure if the `DIRECTION` is a must tho, but it would be nice to have. Also, I'm not sure if `SHAPE ABUTMENT ;` should be added as well.

## Actual Behavior

In all cells of all libraries, VNB and VPB pins are missing the USE and DIRECTION sections.
i.e.:
```
PIN VPB
    PORT
      LAYER nwell ;
        RECT -0.190000 1.305000 2.950000 2.910000 ;
    END
  END VPB
```

## Steps to Reproduce the Problem

1. Check the .magic.lef file of any cell when you clone any of the submodules.

## Specifications

- Version: f3ed3d715481735d2768f5bf51c7c28164020c6e

*Note: the purpose of this issue is traceability.

RTimothyEdwards, 2020-10-20:
I do not think that "SHAPE ABUTMENT" should be used, because the labels in the LEF are not extending to the cell bounding box edges, so I'm not sure what tools would do with that.  The regular power rails abut, which I think should be okay.  But if any tool behavior indicates otherwise, I would like to know about it.

agorararmard, 2020-10-21:
So far no tool indicated the need for `SHAPE ABUTMENT`. So, I agree, it shouldn't be added.

mithro, 2020-10-24:
I'm currently assuming this affects all the `sky130_fd_sc_*` cells?

agorararmard, 2020-10-26:
@mithro: I reviewed the submodules now. The issue exists in sky130_fd_sc_hd, sky130_fd_sc_hdll, and sky130_fd_sc_ls.

So, it doesn't exist in sky130_fd_sc_hs and sky130_fd_sc_ms, sky130_fd_sc_hvl, and sky130_fd_sc_lp

mithro, 2020-10-28:
Does this look right?
```diff
diff --git a/libraries/sky130_fd_sc_hd/v0.0.1/cells/a22oi/sky130_fd_sc_hd__a22oi_4.magic.lef b/libraries/sky130_fd_sc_hd/v0.0.1/cells/a22oi/sky130_fd_sc_hd__a22oi_4.magic.lef
index 481ecf9..ff1bc43 100644
--- a/libraries/sky130_fd_sc_hd/v0.0.1/cells/a22oi/sky130_fd_sc_hd__a22oi_4.magic.lef
+++ b/libraries/sky130_fd_sc_hd/v0.0.1/cells/a22oi/sky130_fd_sc_hd__a22oi_4.magic.lef
@@ -77,7 +77,7 @@
     END
   END Y
   PIN VGND
-    DIRECTION INPUT ;
+    DIRECTION INOUT ;
     SHAPE ABUTMENT ;
     USE GROUND ;
     PORT
@@ -86,7 +86,7 @@
     END
   END VGND
   PIN VNB
-    DIRECTION INPUT ;
+    DIRECTION INOUT ;
     USE GROUND ;
     PORT
       LAYER pwell ;
@@ -94,7 +94,7 @@
     END
   END VNB
   PIN VPB
-    DIRECTION INPUT ;
+    DIRECTION INOUT ;
     USE POWER ;
     PORT
       LAYER nwell ;
@@ -102,7 +102,7 @@
     END
   END VPB
   PIN VPWR
-    DIRECTION INPUT ;
+    DIRECTION INOUT ;
     SHAPE ABUTMENT ;
     USE POWER ;
     PORT
```

```lef
# Copyright 2020 The SkyWater PDK Authors
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     https://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#
# SPDX-License-Identifier: Apache-2.0

VERSION 5.7 ;
  NOWIREEXTENSIONATPIN ON ;
  DIVIDERCHAR "/" ;
  BUSBITCHARS "[]" ;
MACRO sky130_fd_sc_hd__a22oi_4
  CLASS CORE ;
  FOREIGN sky130_fd_sc_hd__a22oi_4 ;
  ORIGIN  0.000000  0.000000 ;
  SIZE  7.820000 BY  2.720000 ;
[... 134 lines cut ...]
      RECT 2.905000  2.635000 3.075000 2.805000 ;
      RECT 3.365000 -0.085000 3.535000 0.085000 ;
      RECT 3.365000  2.635000 3.535000 2.805000 ;
      RECT 3.825000 -0.085000 3.995000 0.085000 ;
      RECT 3.825000  2.635000 3.995000 2.805000 ;
      RECT 4.285000 -0.085000 4.455000 0.085000 ;
      RECT 4.285000  2.635000 4.455000 2.805000 ;
      RECT 4.745000 -0.085000 4.915000 0.085000 ;
      RECT 4.745000  2.635000 4.915000 2.805000 ;
      RECT 5.205000 -0.085000 5.375000 0.085000 ;
      RECT 5.205000  2.635000 5.375000 2.805000 ;
      RECT 5.665000 -0.085000 5.835000 0.085000 ;
      RECT 5.665000  2.635000 5.835000 2.805000 ;
      RECT 6.125000 -0.085000 6.295000 0.085000 ;
      RECT 6.125000  2.635000 6.295000 2.805000 ;
      RECT 6.585000 -0.085000 6.755000 0.085000 ;
      RECT 6.585000  2.635000 6.755000 2.805000 ;
      RECT 7.045000 -0.085000 7.215000 0.085000 ;
      RECT 7.045000  2.635000 7.215000 2.805000 ;
      RECT 7.505000 -0.085000 7.675000 0.085000 ;
      RECT 7.505000  2.635000 7.675000 2.805000 ;
  END
END sky130_fd_sc_hd__a22oi_4
END LIBRARY
```

rovinski, 2020-10-28:
`INPUT` --> `INOUT` seems good to me. It is consistent with other PDKs.

agorararmard, 2020-10-28:
@mithro: Looks good to me!

mithro, 2020-10-29:
@RTimothyEdwards -- What values for the `USE` / `DIRECTION` are correct?

mithro, 2020-10-29:
IE -- https://github.com/RTimothyEdwards/open_pdks/pull/44#pullrequestreview-519219234

mithro, 2020-10-30:
This should be fixed as https://github.com/google/skywater-pdk/pull/196 is now merged.

PR #196 (2020-10-30): Bug fixes to the standard cell libraries.
Three fixes applied to all standard cell libraries;
 * lef: Fixing VNB/VPB properties in .magic.lef files, fixes #172.
 * verilog: Fixing power pins usage in non-powerpin mode, fixes #181.
 * cdl: Fixing missing terminals, fixes #194.

Updating submodules on 2020-10-29 19:19:47 UTC

 - Updating [`sky130_fd_sc_hd` latest](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.2-40-g836b7fa01..v0.0.2-46-g8d095dc6f) to v0.0.2-46-g8d095dc6f
 - Updating [`sky130_fd_sc_hd` v0.0.1](https://foss-eda-tools.googlesource.com/skywater-pdk [... cut]
commit 63fe6bd7 (2020-10-30): Bug fixes to the standard cell libraries.

Three fixes applied to all standard cell libraries;
 * lef: Fixing VNB/VPB properties in .magic.lef files, fixes #172.
 * verilog: Fixing power pins usage in non-powerpin mode, fixes #181.
 * cdl: Fixing missing terminals, fixes #194.

Updating submodules on 2020-10-29 19:19:47 UTC

 - Updating [`sky130_fd_sc_hd` latest](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.2-40-g836b7fa01..v0.0.2-46-g8d095dc6f) to v0.0.2-46-g8d095dc6f
 - Updating [`sky130_fd_sc_hd` v0.0.1](https://foss-eda-tools.googlesource.com/skywater-pd [... cut]
closed 2020-10-30 by mithro