google/skywater-pdk #178: Bad line in `sky130_fd_sc_hd__lpflow_inputisolatch.functional.pp.v`

RTimothyEdwards, 2020-10-14:
Library: sky130_fd_sc_hd
Cell: lpflow_inputisolatch
File: sky130_fd_sc_hd__lpflow_inputisolatch.functional.pp.v
Line: 57

Line is:   "wire 1'b0 ;"  which is illegal verilog syntax.  It seems to have been caused by code that corrects the functional verilog.  "notifier" has been replace with "1'b0", but that should NOT be specified as a wire.

Solution:  Remove this line (or fix the scripts not to generate it in the first place).

mithro, 2020-11-01:
This looks like it has been fixed?

[Line 57 reads;](https://cs.opensource.google/skywater-pdk/sky130_fd_sc_lp/+/master:cells/inputisolatch/sky130_fd_sc_lp__inputisolatch.functional.pp.v;l=57?q=inputisolatch.functional.pp.v&ss=skywater-pdk)
```verilog
    wire SLEEP_B_delayed;
```

mithro, 2020-11-01:
Oh wait, I got the wrong cell. It still appears to be an issue;
https://cs.opensource.google/skywater-pdk/sky130_fd_sc_hd/+/master:cells/lpflow_inputisolatch/sky130_fd_sc_hd__lpflow_inputisolatch.functional.pp.v;l=57?q=inputisolatch.functional.pp.v&ss=skywater-pdk

```verilog
`celldefine
module sky130_fd_sc_hd__lpflow_inputisolatch (
    Q      ,
    D      ,
    SLEEP_B,
    VPWR   ,
    VGND   ,
    VPB    ,
    VNB
);

    // Module ports
    output Q      ;
    input  D      ;
    input  SLEEP_B;
    input  VPWR   ;
    input  VGND   ;
    input  VPB    ;
    input  VNB    ;

    // Local signals
    wire buf_Q;
    wire 1'b0 ;

    //                                     Name     Output  Other arguments
    sky130_fd_sc_hd__udp_dlatch$lP_pp$PG$N dlatch0 (buf_Q , D, SLEEP_B, 1'b0, VPWR, VGND);
    buf                                    buf0    (Q     , buf_Q                       );

endmodule
`endcelldefine
```

RTimothyEdwards, 2020-11-09:
@mithro :  Here is a patch file which fixes the issue:

[issue_178.zip](https://github.com/google/skywater-pdk/files/5513328/issue_178.zip)

closed 2020-11-10 by mithro

mithro, 2020-11-10:
This should have been fixed with https://github.com/google/skywater-pdk/pull/231