google/skywater-pdk #180: Power pin definition in Skywater Foundry Standard Cells

tangxifan, 2020-10-14:
The power pins 'VPWR', 'VGND', 'VPB' and 'VNB' are mainly used by physical design tools, which are not needed when doing functional verification.
When developing RTL for a chip, users are supposed not to handle the power pins. 

## Expected Behavior
When flag 'USE_POWER_PINS' is not defined, the Verilog module should not include any power pins.
For example, the [inverter 1x module](https://cs.opensource.google/skywater-pdk/skywater-pdk/+/master:libraries/sky130_fd_sc_ms/latest/cells/inv/sky130_fd_sc_ms__inv_1.v) may contain power pins when 'USE_POWER_PINS' is enabled.

## Actual Behavior
Verilog modules should not contain power pins when 'USE_POWER_PINS' is disabled.

## Steps to Reproduce the Problem

1. Clone the repository
2. Compile by following guidelines
3. Look into the cell example at `libraries/sky130_fd_sc_ms/latest/cells/inv/sky130_fd_sc_ms__inv_1.v`

## Specifications

- Version: latest master
- Platform: CentOS 7

20Mhz, 2020-10-14:
I'd say these are required for Multi Voltage flows, these can be leveraged by flows based on IEEE 1801. Having said that, it may still be useful to have a flag, since most tools may not be prepared to handle these.

tangxifan, 2020-10-15:
@20Mhz No worries. If the modules with idle power pins are needed for a purpose, it is reasonable to keep them. As you said, I would prefer there is another version (which can be enabled with some flags) where the power pins are not included, in order to perform functional verification.

mithro, 2020-10-15:
It should be pretty easy to generate multiple versions which work via either including a specific set of files directly, or setting a specific define and including the generic files.

mithro, 2020-10-27:
Looking at the [`definition.json` file](https://cs.opensource.google/skywater-pdk/skywater-pdk/+/master:libraries/sky130_fd_sc_ms/latest/cells/inv/definition.json) the pins are correctly identified as being power pins, so I'm a bit confused about why the version which used when `USE_POWER_PINS` is not defined has them as ports.

```json
 "ports": [
  [
   "signal",
   "Y",
   "output",
   ""
  ],
  [
   "signal",
   "A",
   "input",
   ""
  ],
  [
   "power",
   "VPWR",
   "input",
   "supply1"
  ],
  [
   "power",
   "VGND",
   "input",
   "supply0"
  ],
  [
   "power",
   "VPB",
   "input",
   "supply1"
  ],
  [
   "power",
   "VNB",
   "input",
   "supply0"
  ]
 ],
```

The [`sky130_fd_sc_ms__inv.functional.v](https://cs.opensource.google/skywater-pdk/skywater-pdk/+/master:libraries/sky130_fd_sc_ms/latest/cells/inv/sky130_fd_sc_ms__inv.functional.v) has the correct set of ports too.

It looks like it is just affecting the strength variant wrapper files of `XXXX_[1234].v` files.

mithro, 2020-10-27:
I have a fix for this but it will probably take a while to push the fixed due to #172

mithro, 2020-10-29:
@tangxifan - I have pushed fixed verilog files to the modules at https://foss-eda-tools.googlesource.com -- See https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd for example. Can you try updating your submodule manually and see if this fixes your problem?

tangxifan, 2020-11-01:
@mithro I have tried the updated Verilog files. The power pins have been removed from the simplified entities. However, I have an issue from the primitive truth table. An example is the [D-type flip-flop](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/dfxbp/sky130_fd_sc_hd__dfxbp.behavioral.v)

You can see that the primitive is instantiated as 

`sky130_fd_sc_hd__udp_dff$P_pp$PG$N dff0 (buf_Q , D_delayed, CLK_delayed, notifier, VPWR, VGND);`,

where the `VPWR` and `GND` are undefined.

In the [primitive Verilog netlist](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/models/udp_dff_p_pp_pg_n/sky130_fd_sc_hd__udp_dff_p_pp_pg_n.v), the truth table of the primitive relies on the value of `VPWR` and `VGND`.

Therefore, simulation results show that the D-type flip-flop cannot operate as expected. I attached a screenshot about the simulation results using Modelsim.

[image]

Note that this problem may be found in other cells whose truth table relies on power pins.

tangxifan, 2020-11-01:
Also I would suggest to re-open the issue #181  where I posted another issue with the default wire type definition.

rovinski, 2020-11-01:
@mithro I think a problem in general is that many cells seem to have
```
// Voltage supply signals
supply1 VPWR;
supply0 VGND;
supply1 VPB ;
supply0 VNB ;
```
defined outside of the `ifdef USE_POWER_PINS` section. Those signals shouldn't be there if power pins aren't defined.

Another issue: some cells define these signals as `input`, whereas the normal convention is to mark them as `inout`

tangxifan, 2020-11-01:
One more comment: Same issue #181 have been seen in all the `.functional.v` netlists.

mithro, 2020-11-01:
@tangxifan - I think #181 reminds me that we should make sure to use separate GitHub issues for each unique problem (otherwise it is easy for issues to get missed).

tangxifan, 2020-11-01:
@mithro I agree. I would like to post more in that issue. But since it has been closed, I am worried that it may be ignored then. If you can reopen it, I can follow up in that issue.