google/skywater-pdk #181: Verilog Compilation Error when using Mentor ModelSim and iVerilog

tangxifan, 2020-10-14:
## Expected Behavior
Mentor ModelSim and iVerilog should compile the standard cell Verilog without errors.

## Actual Behavior
Modelsim complains about the following errors:
`** ** at /skywater-pdk/libraries/sky130_fd_sc_ms/latest/cells/inv/sky130_fd_sc_ms__inv.behavioral.v(39): (vlog-2892) Net type of 'Y' was not explicitly declared.`

iVerilog complains about the following error:
`/skywater-pdk/libraries/sky130_fd_sc_ms/latest/cells/inv/sky130_fd_sc_ms__inv_1.v:34: Include file sky130_fd_sc_ms__inv.v not found`

This can be solved by adding including directory `-I` but will force a long option when running iVerilog.
Users have to include all the cell directory in the command-line.
Ideally, it should be a push-button process. 

## Steps to Reproduce the Problem
For Mentor Modelsim
1. Clone the repository
2. Compile by following guidelines
3. Create a ModelSim project
4. Add the netlist `/skywater-pdk/libraries/sky130_fd_sc_ms/latest/cells/inv/sky130_fd_sc_ms__inv_1.v`
5. Compile

For iVerilog
1. Clone the repository
2. Compile by following guidelines
3. Go to any path outside the cell libraries.
4. `iverilog /skywater-pdk/libraries/sky130_fd_sc_ms/latest/cells/inv/sky130_fd_sc_ms__inv_1.v`

## Specifications

- Version: Mentor QuestaSim v2019.4, iVerilog v10.3
- Platform: CentOS 7

mithro, 2020-10-30:
This should be fixed as https://github.com/google/skywater-pdk/pull/196 is now merged.

closed 2020-10-30 by mithro

tangxifan, 2020-11-01:
@mithro I have tried the updated netlists again. Modelsim still complains the same errors.
I have tried some tweaks in the Verilog netlist and it fixed the errors.
For example, I changed the line 30 at [netlist](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/inv/sky130_fd_sc_hd__inv.behavioral.v#30) to

`\`default_nettype wire`

Beyond the `*.behavioral.v` netlists, other netlists also require such changes.
For example, the [sku130_fd_sc_hd__inv_2.v](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/inv/sky130_fd_sc_hd__inv_2.v#31)

I would suggest to change like this for all the Verilog netlists. Indeed, I see the `\`default_nettype wire` is defined at the end of these netlists, but my understanding is that Modelsim does not accept the late definition.

mithro, 2020-11-01:
> @mithro I have tried the updated netlists again. Modelsim still complains the same errors.
> I have tried some tweaks in the Verilog netlist and it fixed the errors.
> For example, I changed the line 30 [...]

I have created a new issue at https://github.com/google/skywater-pdk/issues/198 to track the default net type issue.

egiacomin, 2020-11-05:
By the way, this `default_nettype statement is not even required. What created the bug was setting it to "none", but there is no need to setting it to "wire" either, just to remove the statement. This is usually what is done in traditional libraries.

PR #196 (2020-10-30): Bug fixes to the standard cell libraries.
Three fixes applied to all standard cell libraries;
 * lef: Fixing VNB/VPB properties in .magic.lef files, fixes #172.
 * verilog: Fixing power pins usage in non-powerpin mode, fixes #181.
 * cdl: Fixing missing terminals, fixes #194.

Updating submodules on 2020-10-29 19:19:47 UTC

 - Updating [`sky130_fd_sc_hd` latest](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.2-40-g836b7fa01..v0.0.2-46-g8d095dc6f) to v0.0.2-46-g8d095dc6f
 - Updating [`sky130_fd_sc_hd` v0.0.1](https://foss-eda-tools.googlesource.com/skywater-pdk [... cut]
commit 63fe6bd7 (2020-10-30): Bug fixes to the standard cell libraries.

Three fixes applied to all standard cell libraries;
 * lef: Fixing VNB/VPB properties in .magic.lef files, fixes #172.
 * verilog: Fixing power pins usage in non-powerpin mode, fixes #181.
 * cdl: Fixing missing terminals, fixes #194.

Updating submodules on 2020-10-29 19:19:47 UTC

 - Updating [`sky130_fd_sc_hd` latest](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.2-40-g836b7fa01..v0.0.2-46-g8d095dc6f) to v0.0.2-46-g8d095dc6f
 - Updating [`sky130_fd_sc_hd` v0.0.1](https://foss-eda-tools.googlesource.com/skywater-pd [... cut]