google/skywater-pdk #183: Modeling problems on sky130_fd_sc_hd liberty files

20Mhz, 2020-10-14:
Documenting 4 issues noted on  sky130_fd_sc_hd. 

1.  Wrong pin used as related_groun_pin.
Synopsys Library Compiler, complain: 
Error: Line 57389, Cell 'sky130_fd_sc_hd__dlclkp_1', pin 'M0', An invalid attribute 'related_ground_pin' is found. (LBDB-27)

```
        pin ("M0") {
            direction : "internal";
            internal_node : "M0";
            related_ground_pin : "VGND";  /* VNB is wrong*/
            related_power_pin : "VPWR";
        }
```

2. Missing pg_pin on level shifter cell
Synopsys Library Compiler, complain: 
Error: Line 82105, Cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1', pg_pin 'VPWR', An invalid attribute 'VNB' is found. (LBDB-27)

```
pg_pin ("VPWR") {
            pg_type : "primary_power";
            related_bias_pin : "VNB"; /* VNB pin is missing*/
            voltage_name : "VPWR";
}
```

3. Missing Standard cell main rail attribute
sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1 is missing std_cell_main_rail : true on pg_pin group for 'VPWR'. This is needed so implementation tools can insert legally without understanding physical view.
 
4. CSS section showing up without table definition. Likely it shouldn't be on sky130_fd_sc_hd__ff_n40C_1v95.lib (may be on other opconds too)
ccsn_dc table is missing: 
 
```
 ccsn_first_stage () {
                    dc_current ("ccsn_dc") {
```

mithro, 2020-10-29:
FYI - The code which generates the liberty files can be found at https://github.com/google/skywater-pdk/blob/master/scripts/python-skywater-pdk/skywater_pdk/liberty.py

mithro, 2020-10-29:
Will need to do more investigation if this needs changes in the output generation or in the source json data.

20Mhz, 2020-11-04:
- For 1, I think you need the change the cell json. As seen here:
[sky130_fd_sc_hd__dlclkp_2__ff_100C_1v65.lib.json](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/dlclkp/sky130_fd_sc_hd__dlclkp_2__ff_100C_1v65.lib.json)
`840:     "related_ground_pin": "VNB",`
- For 2, this is similar but VNB pg_pin is not listed here:
[sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1__ff_100C_1v65.lib.json](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/lpflow_lsbuf_lh_hl_isowell_tap/sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1__ff_100C_1v65.lib.json)
- I think 3 is similar to 2, and should be added to the pg pin attributes.
- Finally 4 may be related to liberty.py, but will have to look further.

20Mhz, 2021-01-05:
As noted in #288 , the fix for 2 ( LS power pin) should actually be using the NWELL pin which is VPB, not VNB. This naming convention is awfully confusing.

glzhou97, 2023-05-10:
I encountered the same problem with issue 1 mentioned here and in #391. May I ask what is the solution for this issue?

sifferman, 2023-12-05:
I am also encountering this issue. Has anyone had success with using Synopsys tools with this PDK?

## Example

I am trying to use sky130 with Design Compiler, but `lc_shell` can't load the `"sky130_fd_sc_hd__tt_025C_1v80.lib"` file.
```
lc_shell version T-2022.03-SP5
```
```bash
lc_shell
```
```tcl
read_lib $SKYWATER_PDK/libraries/sky130_fd_sc_hd/latest/timing/sky130_fd_sc_hd__tt_025C_1v80.lib
write_lib sky130_fd_sc_hd__tt_025C_1v80 -format db -output library.db
```

```
Error: Line 57389, Cell 'sky130_fd_sc_hd__dlclkp_1', pin 'M0', An invalid attribute 'related_ground_pin' is found. (LBDB-27)
Error: Line 57628, Cell 'sky130_fd_sc_hd__dlclkp_2', pin 'M0', An invalid attribute 'related_ground_pin' is found. (LBDB-27)
Error: Line 57867, Cell 'sky130_fd_sc_hd__dlclkp_4', pin 'M0', An invalid attribute 'related_ground_pin' is found. (LBDB-27)
Error: Line 82105, Cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1', pg_pin 'VPWR', An invalid attribute 'VNB' is found. (LBDB-27)
Error: Line 82239, Cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_2', pg_pin 'VPWR', An invalid attribute 'VNB' is found. (LBDB-27)
Error: Line 82373, Cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_4', pg_pin 'VPWR', An invalid attribute 'VNB' is found. (LBDB-27)
Error: Line 82650, Cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_1', pg_pin 'VPWR', An invalid attribute 'VNB' is found. (LBDB-27)
Error: Line 82784, Cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_2', pg_pin 'VPWR', An invalid attribute 'VNB' is found. (LBDB-27)
Error: Line 82918, Cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_4', pg_pin 'VPWR', An invalid attribute 'VNB' is found. (LBDB-27)
Error: Line 164125, Cell 'sky130_fd_sc_hd__sdlclkp_1', pin 'M0', An invalid attribute 'related_ground_pin' is found. (LBDB-27)
Error: Line 164449, Cell 'sky130_fd_sc_hd__sdlclkp_2', pin 'M0', An invalid attribute 'related_ground_pin' is found. (LBDB-27)
Error: Line 164773, Cell 'sky130_fd_sc_hd__sdlclkp_4', pin 'M0', An invalid attribute 'related_ground_pin' is found. (LBDB-27)
```

20Mhz, 2023-12-05:
Yes, check the scripts below, we patched the .libs before compiling:
https://github.com/20Mhz/skywater-pdk/tree/synopsys_pull/vendor/synopsys

RTimothyEdwards, 2023-12-06:
You should be using the repository at https://github.com/efabless/skywater-pdk-libs-sky130_fd_sc_hd, which has been patched.  However, while the incorrect `pg_pin` attributes have been fixed, it appears that the `related_ground_pin` errors are still in the .lib files;  84 instances throughout the .lib files (although fortunately they are restricted to the HD library).  Since those are easy to detect and fix, I will do that now.

RTimothyEdwards, 2023-12-06:
This issue has now been corrected in the Efabless fork.

chithrar14, 2024-12-06:
Has anyone had success with using Synopsys tools with this PDK? How to convert .tlef to .tf? I got the following error when i used this PDK with Synopsys Fusion Compiler. 
Error: No valid parasitics for LATE corner default. Please make sure the net parasitics are setup properly for the design. Related command: set_parasitics_parameters (SCN-1001)
Error: No valid parasitics for EARLY corner default. Please make sure the net parasitics are setup properly for the design. Related command: set_parasitics_parameters (SCN-1001)
How to solve the above issue. Thanks in advance