google/skywater-pdk #187: Import capacitance data from Google Spreadsheet into the PDK

mithro, 2020-10-22:
## Expected Behavior

Capacitance data is available in the PDK ready to go with various tools.

There is some more information on the capacitance data under the [Parasitic Layout Extraction section of the documentation](https://skywater-pdk.readthedocs.io/en/latest/rules/rcx.html#capacitance-values).

## Actual Behavior

Capacitance data is currently stored as raw data in the ["SKY130 (SkyWater PDK) -- Stackup Capacitance Data [public]"](https://docs.google.com/spreadsheets/d/1N9To-xTiA7FLfQ1SNzWKe-wMckFEXVE9WPkPPjYkaxE/edit#gid=1391862914).

The spreadsheet includes;
 - RCX Model Information
 - Metal + Via Layer Process Variation
 - Capacitance table (Basic and Extended Data)

## Open Questions

 - What is the data format needed by the open source tools?
 - What is the data format needed by the closed source tools?
 - How do we get the capacitance data into those formats?

mithro, 2020-10-22:
FYI - @msaligane

msaligane, 2020-10-22:
@mithro I generated the setRC.tcl used by OpenROAD. It is currently in a private repo but will be put in the public one soon. Meanwhile these are the values I am using:
```

# units are fF,kOhm
--
  | set_layer_rc -layer li1 -capacitance 0.217578 -resistance 0.000939039
  | set_layer_rc -layer met1 -capacitance 0.243182 -resistance 0.001218041
  | set_layer_rc -layer met2 -capacitance 0.311007 -resistance 0.000401691
  | set_layer_rc -layer met3 -capacitance 0.339162 -resistance 0.000167980
  | set_layer_rc -layer met4 -capacitance 0.346672 -resistance 0.000133337
  | set_layer_rc -layer met5 -capacitance 315.935 -resistance 0.000003125
  |  
  | set_layer_rc -layer mcon -capacitance 0.070290 -resistance 0.004475656
  | set_layer_rc -layer via -capacitance 0.153021 -resistance 0.003368786
  | set_layer_rc -layer via2 -capacitance 0.142385 -resistance 0.003368786
  | set_layer_rc -layer via3 -capacitance 0.073086 -resistance 0.000376635
  | set_layer_rc -layer via4 -capacitance 0.591704 -resistance 0.000005686
  |  
  | set_wire_rc -layer li1
  | set_wire_rc -layer met1
  | set_wire_rc -layer met2
  | set_wire_rc -layer met3
  | set_wire_rc -layer met4
  | set_wire_rc -layer met5

```

rovinski, 2020-10-22:
> What is the data format needed by the open source tools?

To my knowledge there aren't really any open source extractors currently available. [OpenRCX](https://github.com/The-OpenROAD-Project/OpenRCX) is in the works, but it's a pattern-based extractor that requires input from a 3D field solver. So I'm not sure if there's anything to be done there. Haven't used Magic parasitic extraction, so I'm not sure what type of model it uses.

There are open-source field solvers like FastCap/FasterCap, etc. but I have no experience with them and I'm not sure if they are applicable here.

> What is the data format needed by the closed source tools?

Synopsys tools use TLU / TLU+. Cadence tools use QRC. Calibre tools use xACT. I believe that the only way to generate these formats is from commercial tools, because the formats have 0 public documentation to my knowledge.

One caveat is the ITF format, which is maintained by Synopsys but open-sourced. The spec is available free through their tap-in program. Synopsys offers a way to convert from ITF to TLU using their tools, I'm not sure if there is a path from ITF to QRC or xACT.

I noticed that #185 added an ITF file, but don't think it contains as much information as it can from the linked spreadsheet.

@msaligane noted that OpenROAD-flow uses a setRC.tcl file. This isn't really a firm file format, but merely a collection of tcl commands used to approximate parasitics. It is even less accurate than a cap table, but it is the best that we have at the moment.

I think that creating a proper ITF file might be the best place to start, because the spec is open and it can be validated with commercial tools. I can reach out to the OpenROAD developers and see if they have any thoughts on this.

taylor-bsg, 2020-10-23:
FYI, this looks wrong to me:

> # units are fF,kOhm

> set_layer_rc -layer met5 -capacitance 315.935 -resistance 0.000003125

This capacitance looks very suspicious.

msaligane, 2020-10-23:
@taylor-bsg Yes, thanks and I am aware. It is messing up all the timing.
But the thing is these values are extracted from the new .tlef

I am still trying to debug what is happening. I am planning to use values straight from the shared sheet (above). Any inputs are welcome..

20Mhz, 2020-10-28:
Hi @msaligane, 
Regarding :

> I noticed that #185 added an ITF file, but don't think it contains as much information as it can from the linked spreadsheet.

I created the ITF from the stack diagram at https://skywater-pdk.readthedocs.io/en/latest/rules/assumptions.html. 

Can you highlight some of the gaps you see? From the spreadsheet, I think the relevant tabs would be second, third, since the first is specific to devices and last two are cap values. Are you thinking in lines of creating MAX/MIN ITF files or just checking the values? 

 Let me know if you have specific thoughts.

Thanks,
Ronald

msaligane, 2020-10-29:
Hi @20Mhz 

I am working on a tapeout right now. But I will get back to this soon.

Thanks
Mehdi

rovinski, 2020-10-29:
@20Mhz usually in ITFs there's a lot of information dedicated to variation caused by etching and thickness. So I expect to see tables like `ETCH_VS_WIDTH_AND_SPACING`, `RHO_VS_WIDTH_AND_SPACING`, and `POLYNOMIAL_BASED_THICKNESS_VARIATION`. I suppose the spreadsheet doesn't really have enough info to do anything here. 

Also @mithro it seems like the table is implying this process has 0 variation, which is just not right.

Other missing parameters:
* `CRT1`, `CRT2`, and `T0` - Tables 2 & 3 should provide this info
* `SIDE_TANGENT` - I see a `Licon1 etch angle` variable in the docs, but nothing for other layers. This would be useful to have.

Questions:
* Where did `CAPACITIVE_ONLY_ETCH` values come from? I see those numbers in the diagram but are you sure that's what they represent?

mithro, 2020-10-29:
The ["Metal Layer Process Variation" sheet](https://docs.google.com/spreadsheets/d/1N9To-xTiA7FLfQ1SNzWKe-wMckFEXVE9WPkPPjYkaxE/edit#gid=1666179135) and ["Via Process Variation" sheet](https://docs.google.com/spreadsheets/d/1N9To-xTiA7FLfQ1SNzWKe-wMckFEXVE9WPkPPjYkaxE/edit#gid=1016452309) have some details about the process variation?

rovinski, 2020-10-29:
Those sheets include temperature variation (`C1` & `C2`), but they don't look like they have any process variation. The metal layer sheet has `StdDev` as "0" for every layer, and the via sheet doesn't have anything listed.

mithro, 2020-10-29:
Ahh, looks like you are right.

mithro, 2020-10-29:
Does this info provide anything useful?
```
# Well declarations
well NWELL {}
well PWELL {}

# Diffusion Layers
diffusion P_SOURCE_DRAIN {
   thickness    	 0.12
   resistivity  	 15
}

diffusion N_SOURCE_DRAIN {
   thickness    	 0.12
   resistivity  	 15
}

# Conducting Layers
conductor Poly {
   min_spacing  	 0.21
   min_width    	 0.15
   height       	 0.3262
   thickness    	 0.180
   resistivity  	 48.2
   temp_tc1     	 0.0008916
   temp_tc2     	 8.443e-07
   gate_forming_layer  	 true
   wire_edge_enlargement_r {
     wee_widths       0.15
     wee_spacings     0.21
     wee_adjustments -0.0280
  }
   wire_edge_enlargement_c {
     wee_widths       0.15
     wee_spacings     0.21
     wee_adjustments  0.0
  }
}

conductor li1 {
   min_spacing  	 0.17
   min_width    	 0.17
   height       	 0.9361
   thickness    	 0.100
   resistivity  	 12.8
   temp_tc1     	 0.0006045
   temp_tc2     	 -3.693e-07
   gate_forming_layer  	 false
   wire_edge_enlargement_r {
     wee_widths      0.17
     wee_spacings    0.17
     wee_adjustments 0.0085
  }
   wire_edge_enlargement_c {
     wee_widths      0.17
     wee_spacings    0.17
     wee_adjustments 0
  }
}

conductor met1 {
   min_spacing  	 0.14
   min_width    	 0.14
   height       	 1.3761
   thickness    	 0.36
   resistivity  	 0.125
   temp_tc1     	 0.003179
   temp_tc2     	 3.094e-07
   gate_forming_layer  	 false
   wire_edge_enlargement_r {
[... 291 lines cut ...]

via   via {
   bottom_layer   	 met1
   top_layer      	 met2
   contact_resistance 4.5
   temp_tc1       	 0.001081
   temp_tc2       	 -1.903e-07
}

via   mcon {
   bottom_layer   	 li1
   top_layer      	 met1
   contact_resistance 9.3
   temp_tc1       	 0.001067
   temp_tc2       	 -5.324e-06
}

via   licon {
   bottom_layer   	 Poly
   top_layer      	 li1
   contact_resistance 152
   temp_tc1       	 0.001249
   temp_tc2       	 -6.647e-06
}
```

rovinski, 2020-10-29:
That looks like an ICT file (compatible with Cadence tools). It provides some more information, like `wire_edge_enlargement`. But `wire_edge_enlargement` usually has an array of values for `wee_widths`, `wee_spacings`, and `wee_adjustments`. This file only has one value, which seems weird (See https://free-online-ebooks.appspot.com/enc/14.17/soceUG/Creating_the_ICT_File.html).

Other than that, I don't see any process variation values - only temperature variation.

To be clear, I'm not sure if tools necessarily _need_ this information, but it is almost certainly information that the foundry _has_ and it would be useful to publish.

mithro, 2020-10-29:
I've contacted SkyWater to see what they can provide here.

20Mhz, 2020-11-10:
@rovinski , sorry for late reply, regarding CAPACITIVE_ONLY_ETCH, answer is no, its not clear for me from diagram alone what is the semantic of the marking. I made an assumption and tried to bound its impact by restricting to capacitive effect.

```
The CAPACITIVE_ONLY_ETCH option applies an etch value to the sidewalls of a conductor. A positive value denotes conductor shrink; a negative value denotes conductor expansion. The adjusted conductor width is equal to the drawn width minus twice the etch value.

Use this option instead of the ETCH option to specify that an etch operation is to be used only for capacitance calculations.
```
In the past I requested ICT file on Slack, but it seem it was not available, will review posted file and check what can be updated on ITF.  TODO.

mithro, 2020-11-10:
@20Mhz Please send any requests to tansell@google.com -- that way they won't get lost on slack.

0ena, 2021-02-24:
Hi folks,

sorry to interrupt in your discussion. Did you manage eventually to create a `qrcTechfile` with the capacitances or get one from the foundry?
I would like to try out the Skywater PDK in my Cadence flow, and it is needed for timing signoff. :-)

Thank you in advance for your responses and time.

Kind regards,
Nassos

20Mhz, 2021-03-05:
@0ena, sorry I don't have experience with Cadence flow, but I assume you can use the ICT shown above to create the file you need.

@rovinski , @mithro @msaligane, I finally got some cycles to update the ITF file, it fixes some fundamental flaws/misconceptions from original and adds additional info from the ICT above. Please review the gist below if you can, I'll later  check-in to my fork. 

https://gist.github.com/20Mhz/42838e1667714abd56fadf3fa441c6c0

I translated the wire_edge_enlargement_r directly to RESISTIVE_ONLY_ETCH this time, it seems capacitive was 0 for all layers. If per width/spacing info becomes available please let me know and I can update ITF with ETCH_VS_WIDTH_AND_SPACING instead.

20Mhz, 2021-03-05:
I've built the following image from the ITF file for reference.

[image]