google/skywater-pdk #194: missing terminal in transistors in .cdl files

msaligane, 2020-10-27:
## Expected Behavior

Correct syntax:
```
.SUBCKT sky130_fd_sc_hd__inv_16 A VGND VNB VPB VPWR Y
*.PININFO A:I VGND:I VNB:I VPB:I VPWR:I Y:O
MMIN1 Y A **VGND** VNB nshort m=16 w=0.65 l=0.15 mult=1 sa=0.265 sb=0.265
+ sd=0.28 topography=normal area=0.063 perim=1.14
MMIP1 Y A **VPWR** VPB phighvt m=16 w=1.0 l=0.15 mult=1 sa=0.265 sb=0.265
+ sd=0.28 topography=normal area=0.063 perim=1.14
.ENDS sky130_fd_sc_hd__inv_16
```

## Actual Behavior

In the .cdl files the transistors are missing s terminal. Issue is noticed in hs and hd libs.
The .spice files seems to be correct.
See example:

```
.SUBCKT sky130_fd_sc_hd__inv_16 A VGND VNB VPB VPWR Y
*.PININFO A:I VGND:I VNB:I VPB:I VPWR:I Y:O
MMIN1 Y A VNB nshort m=16 w=0.65 l=0.15 mult=1 sa=0.265 sb=0.265
+ sd=0.28 topography=normal area=0.063 perim=1.14
MMIP1 Y A VPB phighvt m=16 w=1.0 l=0.15 mult=1 sa=0.265 sb=0.265
+ sd=0.28 topography=normal area=0.063 perim=1.14
.ENDS sky130_fd_sc_hd__inv_16
```

msaligane, 2020-10-27:
@mithro @RTimothyEdwards FYI.

mithro, 2020-10-27:
I /think/ this is related to issue https://github.com/google/skywater-pdk/issues/191 which was fixed for `sky130_fd_pr` library in https://github.com/google/skywater-pdk/pull/193

It appears the same needs to be done for the standard cell libraries.

mithro, 2020-10-28:
Does this look right?
```spice
* Copyright 2020 The SkyWater PDK Authors
*
* Licensed under the Apache License, Version 2.0 (the "License");
* you may not use this file except in compliance with the License.
* You may obtain a copy of the License at
*
*     https://www.apache.org/licenses/LICENSE-2.0
*
* Unless required by applicable law or agreed to in writing, software
* distributed under the License is distributed on an "AS IS" BASIS,
* WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
* See the License for the specific language governing permissions and
* limitations under the License.
*
* SPDX-License-Identifier: Apache-2.0
.SUBCKT sky130_fd_sc_hd__a22oi_1 A1 A2 B1 B2 VGND VNB VPB VPWR Y
*.PININFO A1:I A2:I B1:I B2:I VGND:I VNB:I VPB:I VPWR:I Y:O
MMPA0 pndA A1 VPWR VPB pfet_01v8_hvt m=1 w=1.0 l=0.15 mult=1 sa=0.265
+ sb=0.265 sd=0.28 topography=normal area=0.063 perim=1.14
MMPA1 pndA A2 VPWR VPB pfet_01v8_hvt m=1 w=1.0 l=0.15 mult=1 sa=0.265
+ sb=0.265 sd=0.28 topography=normal area=0.063 perim=1.14
MMPB0 Y B1 pndA VPB pfet_01v8_hvt m=1 w=1.0 l=0.15 mult=1 sa=0.265
+ sb=0.265 sd=0.28 topography=normal area=0.063 perim=1.14
MMPB1 Y B2 pndA VPB pfet_01v8_hvt m=1 w=1.0 l=0.15 mult=1 sa=0.265
+ sb=0.265 sd=0.28 topography=normal area=0.063 perim=1.14
MMNA0 Y A1 sndA1 VNB nfet_01v8 m=1 w=0.65 l=0.15 mult=1 sa=0.265
+ sb=0.265 sd=0.28 topography=normal area=0.063 perim=1.14
MMNA1 sndA1 A2 VGND VNB nfet_01v8 m=1 w=0.65 l=0.15 mult=1 sa=0.265
+ sb=0.265 sd=0.28 topography=normal area=0.063 perim=1.14
MMNB0 Y B1 sndB1 VNB nfet_01v8 m=1 w=0.65 l=0.15 mult=1 sa=0.265
+ sb=0.265 sd=0.28 topography=normal area=0.063 perim=1.14
MMNB1 sndB1 B2 VGND VNB nfet_01v8 m=1 w=0.65 l=0.15 mult=1 sa=0.265
+ sb=0.265 sd=0.28 topography=normal area=0.063 perim=1.14
.ENDS sky130_fd_sc_hd__a22oi_1
```

msaligane, 2020-10-28:
Yes, seems like transistors have the 4 terminals now (d g s b)

mithro, 2020-10-29:
@msaligane - Can you take a look over https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/branch-0.0.1%5E%21/ and confirm it looks right? If so, I'll push the changes for the other standard cell libraries.

msaligane, 2020-10-29:
@mithro Looks good to me. Skimmed over a few cells and they look fine. I will probably use the libs again soon for my designs and I can flag if I see any issues.

mithro, 2020-10-30:
This should be fixed as https://github.com/google/skywater-pdk/pull/196 is now merged.

PR #196 (2020-10-30): Bug fixes to the standard cell libraries.
Three fixes applied to all standard cell libraries;
 * lef: Fixing VNB/VPB properties in .magic.lef files, fixes #172.
 * verilog: Fixing power pins usage in non-powerpin mode, fixes #181.
 * cdl: Fixing missing terminals, fixes #194.

Updating submodules on 2020-10-29 19:19:47 UTC

 - Updating [`sky130_fd_sc_hd` latest](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.2-40-g836b7fa01..v0.0.2-46-g8d095dc6f) to v0.0.2-46-g8d095dc6f
 - Updating [`sky130_fd_sc_hd` v0.0.1](https://foss-eda-tools.googlesource.com/skywater-pdk [... cut]
commit 63fe6bd7 (2020-10-30): Bug fixes to the standard cell libraries.

Three fixes applied to all standard cell libraries;
 * lef: Fixing VNB/VPB properties in .magic.lef files, fixes #172.
 * verilog: Fixing power pins usage in non-powerpin mode, fixes #181.
 * cdl: Fixing missing terminals, fixes #194.

Updating submodules on 2020-10-29 19:19:47 UTC

 - Updating [`sky130_fd_sc_hd` latest](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.2-40-g836b7fa01..v0.0.2-46-g8d095dc6f) to v0.0.2-46-g8d095dc6f
 - Updating [`sky130_fd_sc_hd` v0.0.1](https://foss-eda-tools.googlesource.com/skywater-pd [... cut]
closed 2020-10-30 by mithro