google/skywater-pdk #197: Undefined Delay Constants in Functional Verilog Models for Standard Cells

tangxifan, 2020-11-01:
## Expected Behavior
Then functional Verilog netlists should compile without any errors.

## Actual Behavior
QuestaSim spits the following error:

_skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/dfxbp/sky130_fd_sc_hd__dfxbp.functional.v(53): (vlog-2163) Macro `UNIT_DELAY is undefined._

The UNIT_DELAY is supposed to be a constant used in simulation.
PDK should provide a default value, e.g. 1ns, which can be overwritten by users.

## Steps to Reproduce the Problem
1. Checkout PDK and compile
2. Turn on functional model switch by `define FUNCTIONAL 1
3. Compile Verilog netlist `skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/dfxbp/sky130_fd_sc_hd__dfxbp_1.v` 

## Specifications

- Version: QuestaSim 2019.4
- Platform: CentOS

tangxifan, 2020-11-01:
I have tried to tweak a [netlist](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/dfxbp/sky130_fd_sc_hd__dfxbp.functional.v#53) as following:
`sky130_fd_sc_hd__udp_dff$P \#1 dff0 (buf_Q , D, CLK         )`

It compiles fine.

Hope this can help.

mithro, 2020-11-01:
@tangxifan It was/is expected that the user provided the `UNIT_DELAY` define on the command line (normally using something like `-D` type flag) to their simulation tool.

tangxifan, 2020-11-01:
@mithro Thanks for the advice. I have tried to add the constant definition:

_`define UNIT_DELAY 1_

where I assume an _1ns_ unit delay just for testing.

As a result, Modelsim reports another error:

`skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/dfxbp/sky130_fd_sc_hd__dfxbp.functional.v(53): near "1": syntax error, unexpected INTEGER NUMBER.`

Base on this, I tried to tweak the Verilog [netlist](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/dfxbp/sky130_fd_sc_hd__dfxbp.functional.v#53) as follows:

`sky130_fd_sc_hd__udp_dff$P \#`UNIT_DELAY dff0 (buf_Q , D, CLK         );`

Modelsim compiled successfully.

Therefore, I believe the issue is still valid and the netlist contains syntax errors.

mithro, 2020-11-01:
Can you try 
```verilog
`define UNIT_DELAY #1 
```

tangxifan, 2020-11-01:
> Can you try
> 
> ```verilog
> `define UNIT_DELAY #1 
> ```

It works by using this. Hope this can be documented. Thanks for the help.

mithro, 2020-11-01:
@tangxifan - I do think it is probably reasonable to have a default set.