google/skywater-pdk #199: IfDef functional in user defined primitives not

mithro, 2020-11-01:
The functional `ifdef` found in many primitives is not getting correctly converted into a verilog `ifdef`. See the following example where the `['IfDef(functional)', '']` text appears;

https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/models/udp_dff_p_pp_pg_n/sky130_fd_sc_hd__udp_dff_p_pp_pg_n.v#
```
primitive sky130_fd_sc_hd__udp_dff$P_pp$PG$N (
    Q       ,
    D       ,
    CLK     ,
    NOTIFIER,
    VPWR    ,
    VGND
);

    output Q       ;
    input  D       ;
    input  CLK     ;
    input  NOTIFIER;
    input  VPWR    ;
    input  VGND    ;

    reg Q;

    table
     //          D           CLK  NOTIFIER VPWR VGND :  Qt : Qt+1
                 1           (01)    ?      1    0   :  ?  :  1    ; // clocked data
                 0           (01)    ?      1    0   :  ?  :  0    ;
                 1           (x1)    ?      1    0   :  1  :  1    ; // reducing pessimism
                 0           (x1)    ?      1    0   :  0  :  0    ;
                 1           (0x)    ?      1    0   :  1  :  1    ;
                 0           (0x)    ?      1    0   :  0  :  0    ;
                 0            x      ?      1    0   :  0  :  0    ; // Hold when CLK=X and D=Q
                 1            x      ?      1    0   :  1  :  1    ; // Hold when CLK=X and D=Q
                 ?           (?0)    ?      1    0   :  ?  :  -    ;
                 *            b      ?      1    0   :  ?  :  -    ; // ignore edges on data
        // ['IfDef(functional)', '']                 ?            ?      *      1    0   :  ?  :  -    ;
        // ['Else', '']                 ?            ?      *      1    0   :  ?  :  x    ;
        // ['EndIfDef(functional)', '']                 ?            ?      ?      *    ?   :  ?  :  x    ; // any change on vpwr
                 ?            ?      ?      ?    *   :  ?  :  x    ; // any change on vgnd
    endtable
endprimitive
```

garybeihl, 2026-09-13:
Still present five years on, and I think this is worth reopening attention to because the
consequence is larger than it looks. Three notes: the root cause, a third lost row, and what
it does in simulation.

**Root cause is in the `.tsv` -> `.v` generator.** The table source treats those names as
control rows, not data. From
`models/udp_dff_p_pp_pg_n/sky130_fd_sc_hd__udp_dff_p_pp_pg_n.table.tsv`:

```
?	?	*	1	0	:	?	:	-
Else
?	?	*	1	0	:	?	:	x
EndIfDef(functional)
?	?	?	*	?	:	?	:	x	any change on vpwr
?	?	?	?	*	:	?	:	x	any change on vgnd
```

The generator does not recognise `IfDef(...)` / `Else` / `EndIfDef(...)` as directives. It
formats each as a table row and prefixes the emitted line with the repr of the parsed record,
then comments it out.

**A third row is lost, not two.** The `EndIfDef(functional)` prefix lands on the row that
follows it, so `any change on vpwr` is commented out as well. Comparing the `.tsv` with the
generated `.v`, the rows that disappear are:

1. NOTIFIER change -> `-`  (the `FUNCTIONAL` branch)
2. NOTIFIER change -> `x`  (the timing branch)
3. VPWR change -> `x`      (collateral damage from the `EndIfDef` prefix)

Only the VGND row survives.

**Scope.** In the sky130A release from volare
(`b01f6d281922e2185acf13fb855b338bb97ff460`):

```
sky130_fd_sc_hd/verilog/primitives.v    21 commented rows / 7 primitives:
  udp_dff$NSR_pp$PG$N, udp_dff$P_pp$PG$N, udp_dff$PR_pp$PG$N, udp_dff$PS_pp$PG$N,
  udp_dlatch$lP_pp$PG$N, udp_dlatch$P_pp$PG$N, udp_dlatch$PR_pp$PG$N
sky130_fd_sc_hvl/verilog/primitives.v   15 commented rows
```

Non-power-pin cell views are affected too, since the plain view still instantiates the
`_pp_pg_n` UDP with internal `supply1 VPWR` / `supply0 VGND`.

**What it does in simulation.** With no active row covering the NOTIFIER column, a timing-check
violation cannot corrupt the flop through the table. It often still yields `x`, but only via the
UDP default rule for an unmatched input transition -- and the `FUNCTIONAL` branch that was meant
to suppress that has no effect at all. Observed on an SDF-annotated `sky130_fd_sc_hd__dfxtp_1`
with a deliberate 100 ps setup violation: one simulator reports the violation and leaves Q at a
clean 0, another produces `x`. Both are consistent with the table having no NOTIFIER row. The
lost VPWR row is a straightforward missing behaviour.

**Where a fix could go.** The generated `.v` files live in
`google/skywater-pdk-libs-sky130_fd_sc_hd`, which is archived and has issues disabled, and I
could not find the `.tsv` -> `.v` generator in any public repository -- there is no `IfDef`
handling anywhere in `google/skywater-pdk`. So the correction cannot be made where the damage
is. The only live project that ships these files is `open_pdks`, which already carries a
`custom/sky130_fd_sc_hd/` area, though today that adds new `sky130_ef_*` cells rather than
correcting upstream output, and it passes `primitives.v` through unmodified.

Recording the analysis here in case it is useful to the next person who runs into it: if your
gate-level simulation of sky130 never shows a flop going X on a setup or hold violation, this
is why.

RTimothyEdwards, 2026-09-14:
All of your references are outdated.  The repository now exists on the FOSSi Foundation github site;  there is no longer one single repository covering all of the libraries, but each library is in a separate repository.

Please report issues, if they are current and relevant, to https://github.com/fossi-foundation/skywater-pdk-libs-sky130_fd_sc_hd.

"Volare" also no longer exists, as it has been replaced by "ciel" (same system, different name) (and also now owned by the FOSSi Foundation).