google/skywater-pdk #201: netgen setup property circuit specification bug

d-m-bailey, 2020-11-03:
## Expected Behavior
property statements should reference circuit being processed

## Actual Behavior
property statements reference the other circuit

## Steps to Reproduce the Problem

1. fails when doing x-device comparison, but problem can be seen by examining the code.

Example:
```
    if {[lsearch $cells2 $dev] >= 0} {
        permute "-circuit2 $dev" 1 2
        property "-circuit1 $dev" series enable
        property "-circuit1 $dev" series {w critical}
        property "-circuit1 $dev" series {l add}
```
`-circuit1 `should be `-circuit2 `here.

## Specifications

- Version:
- Platform:

mithro, 2020-11-03:
Which file is this found in?

mithro, 2020-11-03:
I can't see anything like this in the PDK -> https://cs.opensource.google/search?q=circuit1&sq=&ss=skywater-pdk

d-m-bailey, 2020-11-03:
Sorry, I was trying to figure out how to create a pull request to fix it. There are several other similar errors in the file.
 
sky130/netgen/sky130_setup.tcl

d-m-bailey, 2020-11-03:
wrong github. should be in open_pdks. 

sorry again.

closed 2020-11-03 by d-m-bailey