google/skywater-pdk #205: Some standard cells with parallel series connections fail netgen LVS

d-m-bailey, 2020-11-04:
CDL has parallel mos in series. layout has series mos in parallel. Logically equivalent, but not topologically equivalent.
https://skywater-pdk.slack.com/archives/C017UA7LEUV/p1604453653008500

## Expected Behavior
```
Circuit 1 contains 8 devices, Circuit 2 contains 8 devices.
Circuit 1 contains 11 nets,    Circuit 2 contains 11 nets.

Netlists match uniquely.

```

## Actual Behavior
```
Circuit 1 contains 8 devices, Circuit 2 contains 10 devices. *** MISMATCH ***
Circuit 1 contains 11 nets,    Circuit 2 contains 12 nets. *** MISMATCH ***

```

## Steps to Reproduce the Problem

1. extract the mag view 
1. run ext2spice
1. compare to cdl

## Specifications

- Version: commit 5cd70ed
- Platform: MacOS

Sample layout will follow

d-m-bailey, 2020-11-04:
a21o_4
[image]

o21a_4
[image]

a21bo_4
[image]

a211o_4
[image]

mithro, 2020-11-04:
As the slack archive might go away;

> **<Mitch Bailey>** Running openlane on my home mac.
> Looking at the LVS results for the sample spm design, I noticed that no devices were extracted from the layout. Everything is black boxes. Is this intentional?
> Is there a way to link the verilog source to lower level CDL and do LVS at the device level? Does the extraction rule file need to be changed to extract devices?
> Unfortunately, currently the cells extracted with magic and and base CDL cells have different number of pins in different order.
> magic: output before power, no bulk terminals
> cdl: output after power, bulk terminals
> For example:
> designs/spm/runs/*/results/magic/spm.spice
> ```spice
> * Black-box entry for subcircuit sky130_fd_sc_hd__and2_4 abstract view
> .subckt sky130_fd_sc_hd__and2_4 A B X VGND VPWR
> .ends
> open_pdks/sky130/sky130A/libs.ref/sky130_fd_sc_hd/cdl/sky130_fd_sc_hd.cdl
> .SUBCKT sky130_fd_sc_hd__and2_8 A B VGND VNB VPB VPWR X
> MMP0 Y A VPB phighvt ...
> MMP1 Y B VPB phighvt ...
> MMIP0 X y VPB phighvt ...
> MMN0 y A VNB nshort ...
> MMN1 sndA B VNB nshort ...
> MMIN0 X y VNB nshort ...
> .ENDS sky130_fd_sc_hd__and2_4
> ```
> *ALSO*: Please note that the CDL is incorrect! The mosfets have only 3 connections. It looks like the source connections are missing.
> The source of MMN0 should be sndA and the drain y.
> 
> 5:34 PM
> **<Mitch Bailey>** #sky130 There are some standard cells that don't pass netgen LVS due to parallel series connections. The spice definition has 2 devices in series each with m=2 meaning each device is actually 2 devices in parallel that are in series, but the layout has 2 sets of series devices that are in parallel.
> The schematic shows 2 nets that won't match.
> The sample layout shows the suggested connections in blue.
> Should these be logged as issues on google/skywater-pdk?

[image]
[image]

mousaq92, 2022-02-03:
Can you tell me how you resolved this issue please?

RTimothyEdwards, 2022-02-03:
I'm not sure it has been resolved.  The CDL is incorrect and should be corrected, and would presumably need to be fixed manually.  I can insert an exception for any file in open_pdks, which would be one way to solve the problem.

mousaq92, 2022-02-04:
Thank you for your response Mr. Tim. I don't think this issue has been resolved. I am trying to reduce the number of LVS mismatches by dividing the top-level into macros. This removed all device mismatches. I have one net mismatch left. This is definitely not the right way to do things, especially for a relatively small design. I will keep checking back here. Hopefully, this will be resolved soon.

d-m-bailey, 2022-02-04:
@mousaq92 For LVS, what I've been doing is using the spice representation of the standard cells instead of the cdl.