google/skywater-pdk #212: RF PMOS Pin Ordering Incorrect

yrrapt, 2020-11-07:
## Expected Behavior
It is convention that PMOS devices order their pins, drain gate source bulk.

This is how the pins are ordered in the non-RF PMOS device spice models and RF and normal NMOS models.

## Actual Behavior
The rf_pfet devices subcircuits pins are ordered incorrectly (bulk drain gate source) which makes them incompatible with normal symbols.

## Steps to Reproduce the Problem

1. Inspect the spice subcircuit declerations for the sky_130_fd_pr__rf_pfet_01v8_.... devices

## Specifications

- Version:  commit de673cb3b58bebfb26401ba1a51144a224cf2cb6
- Platform: N/A

RTimothyEdwards, 2020-11-07:
@mithro insisted on pin alphabetical order as "canonical" over my protestations.  Normally magic would extract a MOSFET, even as a subcircuit call, with the correct pin order.

These could be tracked down and corrected by hand.  Or we could remove the code that forces the alphabetical ordering.  I think the reason that the alphabetical ordering was applied was because magic is ambivalent about which side it chooses as source and drain, and due to the fact that the order of presentation is an enumeration through a hash table, the order is random and changes each time we run the scripts to generate the files, which means that we keep getting excessive changes in the git database.

Probably the best solution would be additional code that forces the usual SPICE ordering for MOSFETs, bipolars, diodes, and such, by a list of devices.  Or we can just correct the git repo by hand.

mithro, 2020-11-07:
@RTimothyEdwards I insisted on a stable "canonical" ordering. I don't really care if it is alphabetical or some other form. It just needs to be **stable**.

RTimothyEdwards, 2020-11-07:
@mithro :  I think that I mentioned once that I could apply the "readspice" script from magic on these, probably after some modifications.  I'll have a look at it.