google/skywater-pdk #223: Varicap documentation

dwarning, 2020-11-08:
We just trying to implement a charge equation approach for capacitors in ngspice. To verify the function with your models for cap_var_hvt and cap_var_lvt I need few information about the expected behavior in voltage range.
Also I did not understand your test appoach in common. Can I conclude from .json file that a value of 1.6fF is the result for a Vcap=1.8V?
Which tool or script is driving the tests? Is there a device manual for the Sky130 process available?

mithro, 2020-11-08:
See https://skywater-pdk.readthedocs.io/en/latest/rules/device-details.html#varactors for more detail.

I believe there might also be a Analog Verilog model floating around somewhere to? @RTimothyEdwards do you recall?

RTimothyEdwards, 2020-11-08:
@mithro :  There is a verilog-A model for the varactor which is in the original skywater-src-nda.  Because I had another SPICE model for the same device from SkyWater, I just substituted that, rather than figure out how to compile the verilog-A model into ngspice, which would require that we provide all the necessary tools for everyone to do that, since there is no concept of a loadable object module in ngspice.  I assume we can provide this file under NDA, and that would be a good idea.

RTimothyEdwards, 2020-11-08:
@dwarning :  The test approach was just a simple script pulled from some discussion board on ngspice;  it is appropriate for linear caps and not so appropriate for varactors.  But it suffices for the main purpose of providing some value as output that allows us to check that we have not broken anything whenever we update the PDK.  The value of 1.6fF is the value calculated at whatever conditions the testbench has set on that device.  The netlist is run through ngspice to get the result.  These results are also helping to check our spectre-to-spice conversion.

dwarning, 2020-11-09:
Dear all,

thank you for providing the requested informations regarding device 
description.

The varactor spice model is not a problem. From ngspice side we have no 
need to substitute the existing model with a verilog-a model at that it 
would be under NDA which can't handled by our open source simulator. In 
principle we can include verilog-a models by the adms compiler.

Meanwhile we found a way to evaluate the actual varactor model with 
ngspice. See attached the c-v characteristic of the lvt 5x5 device.

mithro, 2020-11-10:
@dwarning - We can probably provide the Verilog-A model source under the Apache 2.0 license (not under NDA). We just need a way to verify the file is useful and correct.

dwarning, 2020-11-10:
As I said at the moment I see no need for change. The model work under ngspice. Please see attachement.
[image]
The min-max values are total different to [https://skywater-pdk.readthedocs.io/en/latest/rules/device-details.html#varactors] but this seems buggy or under different size assumption.