google/skywater-pdk #228: Model binning for the MOSFETs does not cover the device lengths used in the standard cell decap cells.

RTimothyEdwards, 2020-11-09:
The micro-binned MOSFET models do not include the dimensions used for the standard cell decaps.

For example, sky130_fd_sc_hd__decap12 has a device sky130_fd_pr__nfet_01v8 that is W=0.55um, L=4.73um.  There are model bins for W=0.55um but the lengths of the bins are at 1.0, 2.0, 4.0, and 8.0 microns.

My suggestion would be to extend the bins to cover the actual sizes.  The behavior of the decaps will not change very much even if the devices are not modeled well.

Here are the devices for the HD library:

Decap suffix    nFET length  nearest bin
---------------------------------------------------
3                     0.59              0.50
4                     1.05              1.00
6                     1.97              2.00
8                     2.89              4.00
12                   4.73              4.00

jeras, 2020-11-09:
Stefan Schippers wrote this comment on slack #analog-design today, I can confirm this removed the errors, but I do not have the knowledge to tell if this is the correct fix or not.

This can be solved by editing file (example for tt corner):
`sky130_fd_pr__pfet_01v8_hvt__tt.corner.spice`
and letting it include
`sky130_fd_pr__pfet_01v8_hvt__tt.pm3.spice`
instead of
`sky130_fd_pr__pfet_01v8_hvt.pm3.spice`
the default included file has 'point-size' binning , as it defines points instead of regions.
the `_tt.pm3.spice` file defines reasonable continuous regions. This change is already done for `nfet_01v8_lvt` if this patch is applied (fixes the temp->temper and tempo coefficients issues in resistors, contribution from @diadatp). Curious enough is the fact that for other transistors (example: `pfet_01v8_lvt`) this `_tt.pm3.spice` is already the default included file.
I can update the patch file to make the change for these `p_hvt` transistors. I usually work with this patched version of the spice models as i was too much bored of these error messages. Needless to say, it must be verified if the accuracy is acceptable when using these binned models outside their 'point-sized' existence regions.

I had to apply a similar change to `cells/nfet_01v8/sky130_fd_pr__nfet_01v8__tt.corner.spice` to remove the other half of errors. Again I don't know if this is correct.

RTimothyEdwards, 2020-11-09:
@mithro :  Please see the patch file at the end of the posts for issue #171.  It also corrects this issue.

closed 2020-11-11 by mithro