google/skywater-pdk #238: Rectangular Vias allowed?

thesourcerer8, 2020-11-12:
## Expected Behavior
According to https://skywater-pdk.readthedocs.io/en/latest/rules/periphery.html#via it says that Min and Max W and L must be 0.150u 
## Actual Behavior
Magic does not give any DRC violations on rectangular vias

RTimothyEdwards, 2020-11-12:
This is not an error.  Magic drawn contacts are like "via generate" areas in a LEF technology definition;  they define where contact cuts will be placed.  The contact cuts themselves are generated in that area by algorithm.

RTimothyEdwards, 2020-11-12:
@mithro :  This is not even a skywater-pdk issue;  it is a magic usage issue.

thesourcerer8, 2020-11-13:
So if I want an array of vias, I just have to paint a rectangle in .mag and .gds where that array should be ?