google/skywater-pdk #240: conda environment.yml may have unnecessary dependencies

d-m-bailey, 2020-11-13:
It appears that the macos versions of `netlistsvg` and `yosys` are not available via `conda`.
This results in an error when executing `make sky130_fd_sc_hd`
It also appears that `netlistsvg` and `yosys` may not be actually required by any of the programs called by the makefile. 
If this is the case, `netlistsvg` and `yosys` can be removed from `skywater-pdk/environment.yml`

## Expected Behavior
`make sky130_fd_sc_hd` should work in macos.

## Actual Behavior
```
Solving environment: failed

ResolvePackageNotFound: 
  - netlistsvg
  - yosys
```

## Steps to Reproduce the Problem

1. Open macos terminal
1. clone the `skywater-pdk` repo
1. from `skywater-pdk` directory, `make sky130_fd_sc_hd`

## Specifications

- Version: openlane 0fec8c8, skywater-pdk dae7a85
- Platform: macos Catalina

mithro, 2020-11-13:
This should be fixed by https://github.com/google/skywater-pdk/pull/222

Netlistsvg (and thus Yosys) is only needed for building some parts of the documentation.

BTW There should be MacOS X packages of Yosys at https://github.com/litex-hub/litex-conda-eda/tree/master/syn/yosys

d-m-bailey, 2020-11-13:
@mithro I downloaded and installed netlistsvg and yosys, but that wasn't the problem.

It appears that conda tries to find a macos version in the cloud (or wherever it looks) and couldn't find it.

Good to hear that this should be fixed soon.