google/skywater-pdk #250: Device models are not working.

mabrains, 2020-11-19:
We are facing several issues with the device models. We have summarized some of them in the following repo:
https://github.com/mabrains/sky130_analog_testcases/tree/master PFET Vth is incorrect most likely.

## Steps to Reproduce the Problem
All test cases are in the following repo:
https://github.com/mabrains/sky130_analog_testcases/tree/master

We will add the testcases later.

mithro, 2020-11-19:
@mabrains Could you explain the issue in words?

mabrains, 2020-11-19:
We are seeing a significant increase in Vth for pfet with the increase of W of the transistor. Which doesn't makes sense.

mabrains, 2020-11-19:
We are using ng-spice. Please note that we don't have access to paid simulators.

mabrains, 2020-11-19:
Please check the following notebook:
https://github.com/mabrains/sky130_analog_testcases/blob/master/pfet_vth_issue/run_tests.ipynb

mabrains, 2020-11-19:
@mithro Please take a look.

RTimothyEdwards, 2020-11-21:
@mabrains :  This looks like a trend through the data, but do make sure that you are simulating devices only at the specific W and L where they are characterized.  Others have noted significant departure in behavior between the bins.

mabrains, 2020-11-22:
Hi @RTimothyEdwards 

That makes the design very hard. And with this limited number of choices we might not achieve the required goal.

Regards,
Amro

RTimothyEdwards, 2020-11-22:
@mabrains :  I was thinking more along the lines of simulating only at specified points matching the bins for the models, and interpolating from the results.  It's just the models that go wacky outside of the limited bins where they were characterized.  I will get a rejection from SkyWater on devices that are not in the list of "approved" W and L, but those errors can be waived.

RTimothyEdwards, 2020-11-22:
@mabrains :  Still, I cannot say if the problem is really due to simulating device models outside of the characterized points, so this is just speculation.

mabrains, 2020-11-23:
@RTimothyEdwards Just to confirm this is what you refer to as characterized points are at:
https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_pr/+/refs/heads/master/cells/pfet_01v8/sky130_fd_pr__pfet_01v8.bins.csv

RTimothyEdwards, 2020-11-23:
@mabrains :  Yes, that's correct.

mabrains, 2020-11-23:
@RTimothyEdwards Where can I find the measurement data for those sizes?

RTimothyEdwards, 2020-11-23:
@mabrains :  You mean the models?  `libraries/sky130_fd_pr/latest/cells/pfet_01v8/sky130_fd_pr__pfet_01v8.pm3.spice`

RTimothyEdwards, 2020-11-23:
@mabrains :  Note that the lmin/lmax and wmin/wmax of those models are binned very tightly around the specific device W and L.