google/skywater-pdk #281: Liberty Timing file generation is very slow

robtaylor, 2020-12-21:
## Expected Behavior
`make timings` takes a sensible amount of time

## Actual Behavior
`make timings` takes 8 hours on my laptop..

## Steps to Reproduce the Problem

1. `make timings`
1.
1.

## Specifications

- Version: 3d7617a1acb92ea883539bcf22a632d6361a5de4
- Platform: Mac OSX Bug Sur 2016 Macbook

mithro, 2020-12-21:
@robtaylor - Just FYI, `make timings` takes around ~15 minutes on my desktop machine. It might be worth figuring out why there is such a speed difference.

robtaylor, 2020-12-22:
Are you generally running  -j ?

mithro, 2020-12-22:
Yes.

robtaylor, 2020-12-22:
Thats probably it. My lappy only has 4 cores and the code is very single-threaded. We could do a little io/cpu profiling to check..