google/skywater-pdk #285: Minimum Enclosed Area Rule is not covered by the technology LEF for all std cell libraries

agorararmard, 2020-12-23:
## Expected Behavior

For each Technology LEF for each STD library, there should exist a specification for the minimum enclosed area rule: "Min area of metal2 holes > 0.14um^2" and "Min area of metal1 holes > 0.14um^2" this could be achieved by adding the following line:
```
MINENCLOSEDAREA 0.14 ;
```
in each technology LEF in `layer met1` and `layer met2`.

For example for the sky130_fd_sc_hd library it should be added here: https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/tech/sky130_fd_sc_hd.tlef#110 and here: https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/tech/sky130_fd_sc_hd.tlef#152

## Actual Behavior

"Min area of metal2 holes > 0.14um^2" and "Min area of metal1 holes > 0.14um^2" are not covered by the techlef.

https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/tech/sky130_fd_sc_hd.tlef

`MINENCLOSEDAREA 0.14 ;` is missing from the tlef files for all the std libraries:
- sky130_fd_sc_hd
- sky130_fd_sc_hdll
- sky130_fd_sc_hs
- sky130_fd_sc_ms
- sky130_fd_sc_ls

## Steps to Reproduce the Problem

1. Check the techlef of any of the std cells.

## Specifications

- Version:
- Platform:

agorararmard, 2020-12-23:
@mithro: The following gist contains a patch for each of https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/ and https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hdll/ if you apply the patches this issue will be resolved.

https://gist.github.com/agorararmard/d117c1967449b5a78570a4296119ccc5

antonblanchard, 2021-01-14:
I can confirm this fixes the issue for me.

RTimothyEdwards, 2021-03-05:
Corrected in pull request #3 to https://github.com/efabless/skywater-pdk-libs-sky130_fd_sc_hd fork, which automatically updates the upstream repos.  Note that the HDLL library requires a separate pull request, so leave this issue open until that repository has also been updated.

msaligane, 2021-03-31:
@RTimothyEdwards It doesn't sound like the upstream repos are being updated. https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/tree/master/tech

Am I missing something?

RTimothyEdwards, 2021-03-31:
@msaligane :  They aren't.  There was supposed to be a system in place to allow pull requests on the library repositories.  Ask  @mithro about it. . .

RTimothyEdwards, 2021-03-31:
@msaligane :  The upstream merge is still a manual process.  I think that Amr Gouhar (@agorararmard) can do that.

msaligane, 2021-04-01:
Thanks @RTimothyEdwards and @agorararmard . BTW, is the MINENCLOSEDAREA needed in the other libs such as `sky130_fd_sc_hs`. If so, then we need to update the other `tlef` files. Thoughts?

RTimothyEdwards, 2021-04-01:
@msaligane :  The other libraries have it.  It was just missing in HD and HDLL.