google/skywater-pdk #286: Missing leading '*' in line 16 of sky130_fd_pr\latest\cells\nfet_05v0_nvt\sky130_fd_pr__nfet_05v0_nvt.pm3.spice

holvo, 2020-12-24:
## Expected Behavior
Line 16 should read
* include "sky130_fd_pr__esd_nfet_05v0_nvt.pm3"

## Actual Behavior
Line 16 now is:
include "sky130_fd_pr__esd_nfet_05v0_nvt.pm3"

Because of leading 'i', ngspice tries to add a current source.

## Steps to Reproduce the Problem

1.
1.
1.

## Specifications

- Version: v0.20.1
- Platform: Windows 10

dwarning, 2024-02-05:
This problem still exist!
Please comment line 16 in cells/nfet_05v0_nvt/sky130_fd_pr__nfet_05v0_nvt.pm3.spice. The pm3 file is not available. Whatever meaning it has.

RTimothyEdwards, 2024-02-05:
@dwarning :  You should not be using the Google version of the respository, which is not being updated or maintained.  You should be using the Efabless fork of the repository, which has corrected this and several hundred other errors.  The open_pdks installer uses the Efabless fork of sky130_fd_pr, so those corrections are in all installed PDKs derived from open_pdks, including the pre-compiled volare PDKs.

dwarning, 2024-02-05:
Thanks Tim,
I want only simulate. Is there a way to disable all the tool dependencies like magic etc. in configure step of open_pdks?
I tried ngspice simulation with the pure efabless clone but with that the relative pathes are wrong by using skywater-pdk-libs-sky130_fd_pr/combined_models/sky130.lib.spice :

`******
** ngspice-42+ : Circuit level simulation program
** Compiled with Sparse Direct Linear Solver
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Copyright 2001-2023, The ngspice team.
** Please get your ngspice manual from https://ngspice.sourceforge.io/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Wed Jan 24 19:57:01 UTC 2024
******
Error: Could not find include file ../../../libs.ref/sky130_fd_pr/spice/sky130_fd_pr__diode_pd2nw_11v0.model.spice
Error: Could not find include file ../../../../libs.ref/sky130_fd_pr/spice/sky130_fd_pr__esd_nfet_01v8__tt.corner.spice
Error: Could not find include file ../../../../libs.ref/sky130_fd_pr/spice/sky130_fd_pr__rf_nfet_01v8_b__tt.corner.spice
`

RTimothyEdwards, 2024-02-05:
@dwarning :  I think this is my omission;  the "all.spice" file should be pointing to directories in the repository, not the installation, for the express purpose of doing exactly what you are attempting to do, which is to run simulation out of the repository without an installation.  I will correct it.

RTimothyEdwards, 2024-02-05:
@dwarning:  Please pull the latest commit and try again.  Let me know if the issue has not been resolved.

dwarning, 2024-02-06:
@RTimothyEdwards: Thanks, it is much better now - I can simulate with git clone from efabless only.

Using **skywater-pdk-libs-sky130_fd_pr/combined_models/sky130.lib.spice" tt** a small quirk left over:
`Error: Could not find include file ../../cells/cap_vpp_02p9x06p1_m1m2m3m4_shieldl1_fingercap2/sky130_fd_pr__cap_vpp_02p9x06p1_m1m2m3m4_shieldl1_fingercap2.model.spice
Error: Could not find include file ../../cells/cap_vpp_02p9x06p1_m1m2m3m4_shieldl1_fingercap2/sky130_fd_pr__cap_vpp_02p9x06p1_m1m2m3m4_shieldl1_fingercap2.model.spice
.
.
`

Using **skywater-pdk-libs-sky130_fd_pr/models/sky130.lib.spice" tt** fails:
`Error: Could not find include file ../../cells/nfet_05v5/sky130_fd_pr__npn_05v5__t.corner.spice
Error: Could not find include file ../../cells/nfet_05v5/sky130_fd_pr__npn_05v5__t.corner.spice

Note: Compatibility modes selected: hs a

Circuit: nmos noise voltage test

option SCALE: Scale is set to 1e-06 for instance and model parameters
Error: unknown subckt: xm1 g g s 0 sky130_fd_pr__nfet_01v8_lvt w=10 l=1.0
    Simulation interrupted due to error!
`
As I understood the last case is not actual anymore.

RTimothyEdwards, 2024-02-06:
@dwarning :  Thanks for the thorough testing and finding those last (I hope) errors.  I just fixed both in the repository.

dwarning, 2024-02-06:
Now it is clean - not any missing notice. Thank you!
My noise evaluations with nfet_01v8 and pfet_01v8 show reasonable results.
[image]
[image]

RTimothyEdwards, 2024-02-06:
@dwarning:  Good confirmation, and thanks again!