google/skywater-pdk #288: Incorrect Bias Pin Relation in sky130_fd_sc_hd Generated Liberty Files

stdavids, 2020-12-26:
## Expected Behavior

For a standard unbiased CMOS gate, the body pin of pfet transistors are usually tied to power and the body pin of nfet transistors are usually tied to ground. In sky130, VNB and VPB the net names for the body pins for nfet and pfet transistors respectively, as can be seen in the spice model for the [sk130_fd_sc_hd library's](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd) inv_1 cell:

https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/inv/sky130_fd_sc_hd__inv_1.spice#19
```
X0 VGND A Y VNB sky130_fd_pr__nfet_01v8 w=650000u l=150000u
X1 VPWR A Y VPB sky130_fd_pr__pfet_01v8_hvt w=1e+06u l=150000u
```

Shown here, VNB is the body connection for sky130_fd_pr__nfet_01v8 (nfet) and VPB is the body connection for sky130_fd_pr__pfet_01v8_hvt (pfet).

Therefore, inside the [sk130_fd_sc_hd library's](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd) generated lib files I expected to see the the related bias pin for VPWR be VPB, and the related bias pin for VGND be VNB.

## Actual Behavior

What I am actually seeing is the opposite, where VPB is being related to ground and VNW is being related to power.

https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/inv/sky130_fd_sc_hd__inv_1__tt_025C_1v80.lib.json#17
```
"pg_pin,VGND": {
    "pg_type": "primary_ground",
    "related_bias_pin": "VPB",
    "voltage_name": "VGND"
  },
```

https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/inv/sky130_fd_sc_hd__inv_1__tt_025C_1v80.lib.json#32
```
"pg_pin,VPWR": {
    "pg_type": "primary_power",
    "related_bias_pin": "VNB",
    "voltage_name": "VPWR"
  },
```

Here the related_bias_pin for VGND is VPB and the related_bias_pin for VPWR is VNB which is the opposite of what I expected. I have shown this for inv_1 however this swap holds true for many (maybe all?) of the cells in this library.

## Steps to Reproduce the Problem

Visually inspect the links provided.

## Specifications

- Version: [commit ac7fb61f06e6470b94e8afdf7c25268f62fbd7b1](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master)
- Platform: [sk130_fd_sc_hd](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd)

msaligane, 2020-12-28:
Hi @stdavids 
I was confused as well: usually VNW is the pin connected to Nwell but here it is called VPB. Similarly, VPW is the pin for Pwell, and here it is VNB. So you are right, they are inverted in the .lib. I have corrected this locally but hopefully, @mithro / @RTimothyEdwards could fix that here.

20Mhz, 2020-12-29:
Seems related to:  #183

mithro, 2020-12-29:
@RTimothyEdwards - Can you confirm?

RTimothyEdwards, 2020-12-30:
@mithro :  Yes, I can confirm.  Apparently every single liberty file is incorrect.  "related_bias_pin" is swapped.  It's wrong in the .lib.json files.  I don't know the full path from the skywater-src-nda;  "related_bias_pin" is not in skywater-src-nda, so it gets inserted somewhere in the build flow.

robtaylor, 2021-04-02:
@msaligane Could you share your local fix?

msaligane, 2021-04-02:
@robtaylor see attached
[sky130_fd_sc_hd__tt_025C_1v80.zip](https://github.com/google/skywater-pdk/files/6251392/sky130_fd_sc_hd__tt_025C_1v80.zip)

thesukantadey, 2021-04-05:
@robtaylor did you write any script to fix it? Or what procedure did you follow?

robtaylor, 2021-04-05:
I've got a script attached here, unzip and execute in `libraries/sky130_fd_sc_hd/latest`  with `git grep -l "related_bias_pin" | xargs -n1 fixup.sh` 

[fixup.sh.gz](https://github.com/google/skywater-pdk/files/6261029/fixup.sh.gz)

Also my tweaked repo can be found at https://github.com/robtaylor/skywater-pdk-libs-sky130_fd_sc_hd/tree/fixes/288

msaligane, 2021-04-08:
This seems to work however, there is an issue with a few cells: `'sky130_fd_sc_hd__dlclkp_1'`, `dlclkp` and `sdlclkp `cells 

` Error: Line 57389, Cell 'sky130_fd_sc_hd__dlclkp_1', pin 'M0', An invalid attribute 'related_ground_pin' is found.` 

I fixed this using 2 different solutions:
 - change     `related_ground_pin : "VNB";` to   `  related_ground_pin : "VGND";`

current:
```
pin ("M0") {
    direction : "internal";
    internal_node : "M0";
    **related_ground_pin : "VNB";**
    related_power_pin : "VPWR";
}
```
What works:
```
pin ("M0") {
    direction : "internal";
    internal_node : "M0";
    **related_ground_pin : "VGND";**
    related_power_pin : "VPWR";
}
```

The other way that would fix the M0 pin issue and I think it is the original way of declaring `pg_pin(VNB)`:
```
pg_pin (VNB) {
    pg_type : "primary_ground";
    voltage_name : "VNB";
}
```
after the fix discussed above:

```
pg_pin ("VNB") {
    pg_type : "pwell";
    physical_connection : "device_layer";
    voltage_name : "VNB";
}
```
@robtaylor I hope this makes sense..

robtaylor, 2021-04-09:
@msaligane any thoughts on which is the most 'correct' approach?

robtaylor, 2021-04-09:
tweaked script for 1st approach:
[fixup.sh.gz](https://github.com/google/skywater-pdk/files/6287432/fixup.sh.gz)
and tweaked repo again here: https://github.com/robtaylor/skywater-pdk-libs-sky130_fd_sc_hd/tree/fixes/288

RTimothyEdwards, 2021-04-09:
@robtaylor :  Definitely "VNB" should be "pwell" and not "primary_ground", and all "related_ground_pin" entries should be "VGND".

robtaylor, 2021-04-09:
Thank you @RTimothyEdwards, it's good to know i'm on the right path - I'm pretty much flying blind here! 

When this is working, what's the best way to submit a PR? - against https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd ?

RTimothyEdwards, 2021-04-09:
@robtaylor :  There is, unfortunately, as yet no way to submit a pull request.  Only @mithro knows when that will happen.