google/skywater-pdk #293: DRC error in sky130_fd_sc_hd__buf_16 cell

jack-melchert, 2021-01-07:
## Expected Behavior
No DRC error in buf_16 cell.

## Actual Behavior
Magic reports a DRC error for sky130_fd_sc_hd__buf_16.gds.

DRC error is licon.8a: poly overlap of poly contact < 0.08um in one direction.
 
[image]

This poly contact (box with black border) is only 0.06 microns from the edge of the poly layer (shown in red). The corresponding DRC errors are shown in the dotted white area.

## Steps to Reproduce the Problem

1. Open magic in same directory with .magicrc file found here: https://github.com/RTimothyEdwards/open_pdks/blob/master/sky130/magic/sky130.magicrc. This file has some paths that need to be set to point to the pdk location.
2. Click File -> Read GDS.
3. Locate sky130_fd_sc_hd__buf_16.gds and open.
4. Click Drc -> DRC Find next error.

## Specifications

- Version: Commit 3d7617a1acb92ea883539bcf22a632d6361a5de4
- Platform: sky130_fd_sc_hd

RTimothyEdwards, 2021-01-08:
@mithro :  This is another example to be added to issue #261.  Somebody should go through the standard cells and see how many of them are getting DRC errors flagged in magic.  Also would be useful to check Calibre to see if there is some special rule handling for standard cells that isn't in the documentation (this seems unlikely, because I have not yet seen an instance where this error, which is similar in all cases, couldn't be easily fixed just by moving the poly or contact a bit).

egiacomin, 2021-03-17:
@jack-melchert @RTimothyEdwards FY, I just did a DRC check with Calibre and did not get any DRC error on this particular cell.

RTimothyEdwards, 2021-03-17:
@egiacomin :  I had a suspicion that these errors existed because Calibre does not consider them to be errors.  It is not clear to me _why_ Calibre does not consider them to be errors.  I do not see any exceptions to the rule licon.8a for standard cells.  I am not all that familiar with Calibre syntax, so there may be something about the rule that allows this.  The magic rule just follows the rule as stated in the DRC documentation.

egiacomin, 2021-03-27:
@RTimothyEdwards It's not clear to me as well. I haven't played with DRC syntax for some time. I'll give it a try in the coming weeks and let you know if I find anything.

RTimothyEdwards, 2021-03-27:
@egiacomin :  Okay, thanks.  My take on it is that (1) Magic treats it as a violation;  (2) when I measure it, it appears to be a violation, and (3) all of the (apparent) violations that I have seen are quite easy to fix with a single layer adjustment.  So I consider it worthwhile to just modify the layouts so that all tools agree that they meet DRC rules.  On the other hand, I am certainly curious as to why Calibre does not consider these to be errors.