google/skywater-pdk #294: klayout layer properties file

beeflobill, 2021-01-09:
I put together a klayout layer properties file.  So, when trying to use klayout to look at the GDS files we can see the layer names and have a decent color and stipple scheme.  This seemed like a good idea so klayout would be more useful.  (I thought of maybe doing the same thing with magic, but I don't know anything about magic.  So.....)

I used the info in gds_layers.csv to put it together so I wouldn't miss any layers.  I fiddled around with the colors to make it look okay for some random cells in the library.  It's nothing special, but it's better than nothing.

I thought something like this would be useful in the pdk, but where to put it?  I thought maybe in:
     skywater-pdk/libraries/tech

Is there somewhere better?  Does anybody want this?

RTimothyEdwards, 2021-01-10:
Currently, open_pdks (https://github.com/RTimothyEdwards/open_pdks) installs some klayout setup files, but as I do not use klayout much myself, I don't do any development work on the setup files.  There is a very good setup file that is pretty thorough and has DRC rules encoded as well;  I have not incorporated that into open_pdks yet, as I just started working with pulling repositories from outside into the open_pdks build, and I have not gotten around to that one yet.

RTimothyEdwards, 2021-01-10:
@beeflobill :  The repository I mentioned above is here:  https://github.com/laurentc2/SKY130_for_KLayout