google/skywater-pdk #295: Makefile race leads to `.condarc` corruption

tommythorn, 2021-01-22:
## Expected Behavior

No error

## Actual Behavior

```
ConfigurationLoadError: Unable to load configuration file.
  path: /u/skywater-pdk/env/conda/.condarc
  reason: invalid yaml at line 3, column 0
```

## Steps to Reproduce the Problem

1. Follow the "Quick Start" directions of https://github.com/efabless/openlane. 
    I haven't yet been able to reproduce it out and thought it more important to
    capture first what I have found.

## Specifications

- Version: openlane @ c2282d636646797f165013bce5273b0113a54de5 which seems to pull in skywater-pdk @ f6f76f3dc99526c6fc2cfede19b5b1227d4ebde7
- Platform: Ubuntu 21.04 (beta) on a Ryzen 9 3900XT

## Found So Far

The original bug: https://github.com/efabless/openlane/issues/174
Note the "Makefile race" is speculation on my part.  Here's some findings:

The contents of `/u/skywater-pdk/env/conda/.condarc`:
```
envs_dirs:
  - /u/skywater-pdk/env/conda/envs
onda-pkgs
```

I see [only] two write[r]s to this file:
```
60058 openat(AT_FDCWD, "/u/skywater-pdk/env/conda/.condarc", O_WRONLY|O_CREAT|O_TRUNC|O_CLOEXEC, 0666) = 4
60061 openat(AT_FDCWD, "/u/skywater-pdk/env/conda/.condarc", O_WRONLY|O_CREAT|O_TRUNC|O_CLOEXEC, 0666 <unfinished ...>
```
Process 60058 is 
```
60058 execve("/u/skywater-pdk/env/conda/bin/conda", ["/u/skywater-pdk/env/conda/bin/co"..., "config", "--system", "--add", "pkgs_dirs", "/u/skywater-pdk/env/downloads/co"...], 0x56107c417a90 /* 44 vars */ <unfinished ...>
```
and process 60061 is
```
60061 execve("/u/skywater-pdk/env/conda/bin/conda", ["/u/skywater-pdk/env/conda/bin/co"..., "config", "--system", "--add", "envs_dirs", "/u/skywater-pdk/env/conda/envs"], 0x558e51b77af0 /* 44 vars */ <unfinished ...>
```

I'll try to get a cleaner repro.

tommythorn, 2021-01-25:
The issue was the presence of a MAKEFLAGS environment variable set to -j24. This breaks skywater-pdk with isn't parallel make safe. It's fixable but also trivial to workaround, so closing.

mithro, 2021-01-25:
@tommythorn - I think we should either work with `-j` or detect `-j` and error out.