google/skywater-pdk #297: Invalid verilog syntax at  sky130_fd_sc_hd__dlxbn.behavioral.pp.v and sky130_fd_sc_hd__dlxbn.behavioral.v

Manarabdelaty, 2021-01-26:
## Expected Behavior

Verilog views shouldn't have any syntax errors. 

## Actual Behavior

The behavioral model for the `sky130_fd_sc_hd__dlxbn`  has an invalid verilog syntax at `wire 1;`

The issue exists in the following two files: 
https://cs.opensource.google/skywater-pdk/sky130_fd_sc_hd/+/master:cells/dlxbn/sky130_fd_sc_hd__dlxbn.behavioral.v;l=62
https://cs.opensource.google/skywater-pdk/sky130_fd_sc_hd/+/master:cells/dlxbn/sky130_fd_sc_hd__dlxbn.behavioral.pp.v;l=64

## Steps to Reproduce the Problem

Compiling the verilog files using iverilog will flag a syntax error at `wire 1;` 

1.  iverilog sky130_fd_sc_hd__dlxbn.behavioral.v
2.  iverilog sky130_fd_sc_hd__dlxbn.behavioral.pp.v

## Specifications

- Version:
- Platform:

mattvenn, 2021-02-04:
this causes problems in gate level simulation. Work-around currently is to comment out the wire 1 lines.

mattvenn, 2021-02-16:
for my version of the PDK (installed by OpenLANE tagged mpw-one-b), the 2 lines are on 49033 & 49175 of sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v

RTimothyEdwards, 2021-03-05:
Corrected in pull request #4 to https://github.com/efabless/skywater-pdk-libs-sky130_fd_sc_hd fork, which automatically updates the upstream repos.