google/skywater-pdk #301: Beginner Question: Tutorial on NGSPICE simulation

greenscreenflicker, 2021-02-09:
1. Is there a community to assists with questions? Can you provide a link?
2. Is there a tutorial on how to start analog simulation with ngspice?
(Also just a basic inverter would be enough to get to know the tools?)

mithro, 2021-02-09:
You can find the #analog-design channel on the SkyWater PDK Slack channel will be very helpful. You can get an invite to the slack workspace using https://join.skwyater.tools/

You might also find the tutorials from http://www.ZeroToASIC.com interesting too.

You might also find the examples in the `tests` directories under `skywater-pdk/libraries/sky130_fd_pr/latest/cells/<cellname>/` could also be helpful.

We **do** need to improve the beginner documentation around this topic and would love some help in that regard!

greenscreenflicker, 2021-02-09:
@mithro Sorry, I'm afraid the first link does not work for me. 

I will have a look at your proposed documentation soon. Thanks so far!

RTimothyEdwards, 2021-02-09:
@greenscreenflicker :  That link goes down occasionally.  We'll get it back promptly.

mithro, 2021-02-09:
The correct link is https://join.skywater.tools/ (not `https://join.skwyater.tools/`)

holvo, 2021-02-18:
Some links (not directly connected to skywater) are found here: http://ngspice.sourceforge.net/tutorials.html

danchitnis, 2021-03-22:
I managed to get one model working, but still quite not right. Please copy paste the netlist below in [EEsim](https://eesim.dev/) to see the IV curve traces for the [1.8V nfet](https://cs.opensource.google/skywater-pdk/sky130_fd_pr/+/master:cells/nfet_01v8/sky130_fd_pr__nfet_01v8.pm3.spice). (Remember to tick the 'Axis' box)
```plaintext
Basic RLC circuit
.include modelcard.skywater

.param MC_SWITCH=0
.option scale=1E-6

Rg 1 2 680
X1 3 2 0 0 sky130_fd_pr__nfet_01v8
Rd 3 4 100

* Supply
Vid 5 4 DC 0V
Vgb 1 0 DC 0V
Vdd 5 0 DC 3.3V

* This is the analysis
.dc Vdd 0 1.8 0.01 Vgb 0 1.2 0.1

.save i(Vid)

.end
```
The netlist is similar to the [original test](https://cs.opensource.google/skywater-pdk/sky130_fd_pr/+/master:cells/nfet_01v8/tests/sky130_fd_pr__nfet_01v8__nfet_vth.spice) in the Skywater kit. I have manually hacked the [transistor model](https://github.com/danchitnis/EEsim/blob/main/src/sim/models/skywater/mosfet.ts) to avoid complex and nested behaviour for now. The result of the simulation looks similar to the [test results](https://cs.opensource.google/skywater-pdk/sky130_fd_pr/+/master:cells/nfet_01v8/tests/sky130_fd_pr__nfet_01v8_ids_v_vds.svg) in the kit, but there are small differences. This requires quite a bit of understanding of how the models are arranged in the context of this PDK. Overall it requires a fair amount of effort to clean-up the models, and most importantly significant amount of verification to ensure the models are implemented as intended. 

@mithro is there a manual in the original kit describing the naming convention for the models and simulation corners?

@holvo The models in Skywater are binned. It seems to work with ngspice. Can you provide some information on how ngspice interprets the bins of the model so that we can avoid bugs from the ngspice side.

holvo, 2021-03-23:
You should at least specify W and L with your transistor.

`xm02 z a vss vss sky130_fd_pr__nfet_01v8  l=0.15  w=0.495 as=0.131175 ad=0.131175 ps=1.52 pd=1.52`

ngspice-34 looks up W and L given in each X instance (transistor call) and selects the appropriate bin from the given set of model parameters (the SkyWater PDK offers up to 160 bins per model).

This is how my ngspice netlist with a simple inverter looks like:

```
* SkyWater PDK
* simple inverter

.include "D:\Spice_general\skywater-pdk\libraries\sky130_fd_pr\latest\models\corners/tt.spice"
*.include "D:\Spice_general\skywater-pdk\libraries\sky130_fd_pr\latest\models\r+c/res_typical__cap_typical.spice"
*.include "D:\Spice_general\skywater-pdk\libraries\sky130_fd_pr\latest\models\r+c/res_typical__cap_typical__lin.spice"

* the voltage sources: 
Vdd vdd gnd DC 1.8
V1 in gnd pulse(0 1.8 0p 200p 100p 1n 2n)

Xnot1 in vdd gnd out not1

.subckt not1 a vdd vss z
xm01   z a     vdd     vdd sky130_fd_pr__pfet_01v8_hvt  l=0.15  w=0.99  as=0.26235  ad=0.26235  ps=2.51   pd=2.51
xm02   z a     vss     vss sky130_fd_pr__nfet_01v8  l=0.15  w=0.495 as=0.131175 ad=0.131175 ps=1.52   pd=1.52
c3  a     vss   0.384f
c2  z     vss   0.576f
.ends

* simulation command: 
.tran 1ps 10ns 0 10p

.control
run
*set nolegend
plot in out
*rusage all
.endc

.end
```

danchitnis, 2021-03-25:
Thanks Holger, the model assumes defaults values of `w=1 l=1`. It appears that at least for `nfet18` there are two models, one has 63 bins, and the other which is the `tt` corner has 180 bins. Now I am using the `tt` model with `w=1u l=1u` and I am getting matching results with the test results in the PDK.
 
I have created a few [examples](https://github.com/danchitnis/EEsim/blob/main/skywater.md) for anyone who is interested to test and experiment in [EEsim](https://eesim.dev/). I will add more models eventually.
 
@holvo - I am interested to know how ngspice handles the bins in boundary conditions and when there is a gap between the bins?
 
@RTimothyEdwards @mithro - A better organization of the documentation would help to find the information faster. What you refer to as [Design Rules](https://skywater-pdk.readthedocs.io/en/latest/rules.html) is mainly "Process Information", and the actual "Design Rules" for DRC is [Periphery Rules](https://skywater-pdk.readthedocs.io/en/latest/rules/periphery.html). The pages are too long and slow to scroll, perhaps the MD style rendering needs a bit of optimization.

Kreijstal, 2025-02-02:
> You can find the #analog-design channel on the SkyWater PDK Slack channel will be very helpful. You can get an invite to the slack workspace using https://join.skywater.tools/
>
> You might also find [...]

zerotoasic is dead

RTimothyEdwards, 2025-02-03:
You should be able to get to Matt Venn's Zero To ASIC courses through the https://tinytapeout.com site.

holvo, 2025-02-03:
The link is https://www.zerotoasiccourse.com/