google/skywater-pdk #308: Wrong nwell and pwell attributes in .lef files

egiacomin, 2021-03-17:
## Expected Behavior
The .lef file should load without any issue on Innovus.

## Actual Behavior
I am getting an error on Innovus when loading the .tlef file (_skywater-pdk/libraries/sky130_fd_sc_hd/latest/tech/sky130_fd_sc_hd.tlef_):
`**ERROR: (IMPLF-121):	You need to have cut layer after layer 'pwell'.`

## Steps to Reproduce the Problem
This error was "introduced" after the push from [121](https://github.com/google/skywater-pdk/issues/121).

1. Load a design on Innovus with the _skywater-pdk/libraries/sky130_fd_sc_hd/latest/tech/sky130_fd_sc_hd.tlef_ file.
From the lef documentation (http://www.ispd.cc/contests/18/lefdefref.pdf, page 76), a masterslice layer should be followed by a cut layer. In this case, the masterslice layer (pwell/nwell) is followed by a routing layer (li1). 

One workaround to fix this issue (as usually done in some commercial PDK) is to add the appropriate cut layer in the lef file, which would be _licon1_. Ths would be something as follows (the spacing, antenna and other values would have been to be changed, I simply reused the ones from mcon for now to give an example. If anyone wants to chime in, feel free to do so.

```
LAYER poly
	TYPE MASTERSLICE ;
END poly

LAYER licon1
  TYPE CUT ;

  WIDTH 0.17 ;                # Mcon 1
  SPACING 0.19 ;              # Mcon 2
  ENCLOSURE BELOW 0 0 ;       # Mcon 4
  ENCLOSURE ABOVE 0.03 0.06 ; # Met1 4 / Met1 5

  ANTENNADIFFAREARATIO PWL ( ( 0 3 ) ( 0.0125 3 ) ( 0.0225 3.405 ) ( 22.5 408 ) ) ;
  DCCURRENTDENSITY AVERAGE 0.36 ; # mA per via Iavg_max at Tj = 90oC
END licon1
```

## Specifications

- Version: [db2e067](https://github.com/google/skywater-pdk/commit/db2e06709dc3d876aa6b74a5f3893fa5f1bc2a6e)
- Platform: CentOS 7.5, Innovus 20.13.

UmerShahidengr, 2022-04-20:
Does adding a layer in this fashion make the tlef different from the actual process? Will this cause any problems in the fabrication?

takeBrian, 2022-04-20:
Does adding a layer in this fashion make the tlef different from the actual process? Will this cause any problems in the fabrication?

tdene, 2022-06-02:
To emphasize, the current tech-lef files do not meet the official LEF/DEF guidelines because of this issue.