google/skywater-pdk #310: Gate Level Simulation Doesn't Work As Expected for sky130_fd_sc_hd__dfbbn

kalhankoul96, 2021-03-18:
## Expected Behavior
 Gate level simulation works by default. 

## Actual Behavior

Without FUNCTIONAL defined, gate level simulations use uses ".behavioral.pp.v" cells which does not work. 

## Steps to Reproduce the Problem

1. Run a gate level simulation utilizing this cell "https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/blob/master/cells/dfbbn/sky130_fd_sc_hd__dfbbn.behavioral.pp.v".  (Also the non power pins version)
2. This behavioral version leaves several things unconnected and therefore does not work.
3. Works when defining FUNCTIONAL.

## Specifications

- Version: f6f76f3dc99526c6fc2cfede19b5b1227d4ebde7 
- Platform: sky130_fd_sc_hd

RTimothyEdwards, 2021-03-18:
I am not aware of any issues.  Can you provide a specific error message?

kalhankoul96, 2021-03-18:
Sorry for the confusion. The exact issue is that several wires are undriven for example "RESET_B_DELAYED" on line 78.

https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/blob/ac7fb61f06e6470b94e8afdf7c25268f62fbd7b1/cells/dfbbn/sky130_fd_sc_hd__dfbbn.behavioral.pp.v#L78

This causes the flip flop to not work, and then our overall design fails simulation.

RTimothyEdwards, 2021-03-18:
@mithro :  The "specify" section was pulled out of all of the verilog files because iverilog does not handle timing.  However, the $setuphold() statements work like buffers inserted in front of or after the gate.  For example, in the above case, "D" is the gate input, but $setuphold() takes D as input and produces D_delayed;  likewise for the delayed reset, set, and clock.  To make the file iverilog compatible, if the naming is consistent, then it should suffice to remove the suffix "_delayed" from any signal in the verilog files.  I'm not sure if there is any other fallout from doing that, and it needs to be confirmed.

RTimothyEdwards, 2021-03-18:
It is probably worth checking whether or not iverilog handles the signals through the $setuphold() statement, even if it does not handle the actual timing.  A bunch of warning statements out of iverilog would be preferable to a broken behavioral simulation.

KPrabs106, 2021-05-13:
Is there a way we could get the specify blocks with timing checks? It would be very useful to be able to run an sdf annotated gate level simulation

thesourcerer8, 2021-05-30:
We got affected by this issue too. @RTimothyEdwards I would suggest to automatically generate non-delayed verilog versions of the libraries by open_pdks which are suitable for IVerilog (which remove the _delayed and remove any duplicate RESET wires)

RTimothyEdwards, 2021-05-30:
I have no control over this.

thesourcerer8, 2021-05-31:
Ok, I will try to develop it and provide a pull request...