google/skywater-pdk #311: Incorrect ANTENNADIFFAREA in diode cells

tdene, 2021-03-22:
All the diode_2 cell LEFs, such as at https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/blob/master/cells/diode/sky130_fd_sc_hd__diode_2.magic.lef have a non-zero value for ANTENNAGATEAREA . Indeed, the value in question seems to be a duplicate of ANTENNADIFFAREA .

This appears to be incorrect.

The LEF/DEF reference manual defines ANTENNAGATEAREA as such:
"Specifies the gate area, in micron-squared units, to which the pin is connected on a layer."

The diode_2 cells have no poly, and thus no gate area, attached to the DIODE pin, so there either should be no value for ANTENNAGATEAREA , or it should be listed as 0.

egiacomin, 2021-03-24:
That's right. However, there should not be a ANTENNADIFFAREA value either, as this is only for output pins.
Looks like the "non-magic" lef is the correct one.

[https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/blob/master/cells/diode/sky130_fd_sc_hd__diode_2.lef](https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/blob/master/cells/diode/sky130_fd_sc_hd__diode_2.lef)

RTimothyEdwards, 2021-03-24:
@egiacomin :  I strongly disagree with that statement.  The purpose of ANTENNADIFFAREA is to determine when an antenna violation is present and how to go about fixing it.  Any ANTENNAGATEAREA will contribute to a violation, and any ANTENNADIFFAREA will mitigate against it.  It has nothing to do with whether or not it is an output, only that it is diffusion and therefore creates a current path to sink charge buildup during manufacture.

egiacomin, 2021-03-24:
@RTimothyEdwards your point makes sense. But I have personally never seen a .lef file from all the foundries I've worked with specifying the ANTENNADIFFAREA for something else than an output pin, so I guess there might be a reason? I'll try to dig the DEF/LEF documentation and see if I find anything. But if you have any insight, feel free to chime in.

tdene, 2021-03-24:
I would say that the correct way to look at it is that ANTENNADIFFAREA is defined for pins that are connected to diffusion. Typically that is only output pins, as one would expect input pins to be connected to a gate instead.

The exception for the diode cell is due to the fact that the diode's cell single pin acts as both input and output in a way (or neither, really). Since the only point of the diode cell is to fix antenna violations, it is inserted with its single pin connected directly to the target net. There isn't really an input or output, the way I look at it, but rather just a diode connected in parallel.

I have personally seen this same design choice of diode cell used in a commercial standard cell kit, and they indeed listed ANTENNADIFFAREA but no ANTENNAGATEAREA.

(that said, the title of this issue has a typo, my bad, hopefully github will let me edit)

RTimothyEdwards, 2021-03-24:
@egiacomin :  I can confirm for at least one other major foundry's IP that the antenna tie-down cell has the pin declared as an input and also declares the ANTENNADIFFAREA.

msaligane, 2021-03-24:
Hi All

I agree that it should be declared as an input.

This is what I think would be the right LEF description of the hd diode cell as an example:
```
MACRO sky130_fd_sc_hd__diode_2
  CLASS CORE ;
  SOURCE USER ;
  ORIGIN 0 0 ;
  SIZE 0.92 BY 2.72 ;
  SYMMETRY X Y R90 ;
  SITE unithd ;
PIN DIODE
    DIRECTION INPUT ;
    USE SIGNAL ;
    PORT
        LAYER li1 ;
             RECT 0.085 0.255 0.835 2.465 ;
    END
    ANTENNADIFFAREA 0.4347 LAYER met1 ;
    ANTENNADIFFAREA 0.4347 LAYER met2 ;
    ANTENNADIFFAREA 0.4347 LAYER met3 ;
    ANTENNADIFFAREA 0.4347 LAYER met4 ;
    ANTENNADIFFAREA 0.4347 LAYER met5 ;
    ANTENNADIFFAREA 0.4347 LAYER rdl ;
END DIODE
```

This example has been used and confirmed to work in both proprietary tools and openroad.

egiacomin, 2021-03-24:
@RTimothyEdwards Thanks for checking that. I was not aware, so that is good to know. As @tdene mentioned, I believe that having it as input or output would not make any difference. But you are right, in any case, the ANTENNADIFFAREA value should be present.

RTimothyEdwards, 2021-03-24:
@msaligane :  It is hard to decipher the LEF/DEF documentation, and in this case they have no explanation of what the LAYER part of that statement is supposed to be.  The way I usually see it done is to have no LAYER at all, so just "ANTENNADIFFAREA 0.4347 ;".  Specifying any layers above those layers used to make up the pin is meaningless.  The only reason to specify a layer would be if the pin connects to multiple metal layers inside the cell, in which case it is possible that a different diffusion area is seen depending on whether the router connects to one metal layer or another.  Although I would point out that the LEF spec syntax cannot capture the possible complexities of multiple ports.  At any rate, for the vast majority of cells, there is only one port per pin and only one metal to connect to, so just "ANTENNADIFFAREA <value> ;" suffices (and the LEF file is smaller).