google/skywater-pdk #313: Share Innovus map file

egiacomin, 2021-03-27:
I just finished a design and created the Innovus map file by hand to be able to export my .gds and import it sucessfully on Virtuoso. I've checked the _skywater-pdk/vendor/synopsys/PlaceRoute/common_ but none of the provided file were compatible with Innovus.
What would be the best way to share this file? Should I do a pull request and add it to the _skywater-pdk/vendor/synopsys/PlaceRoute/common_ folder?

nguyendao-uom, 2021-04-12:
Hi @egiacomin, I'm also looking for the Innovus map file. Can you share it here? Thanks

egiacomin, 2021-04-12:
Here is the one I uses. Let me know if you run into any issues when using it!

```
# CADENCE Stream layer mapping table for Skywater130nm
# <t_ layerObjectName> <t_ objectType> <n_ layerNum> <n_ datatype>

poly		NET, SPNET, VIA 	66 	20
poly		PIN 							66 	16
licon1	VIA 							66 	44
licon1	PIN 							66 	41
licon1	 NET 							66 	58
li1			NET, SPNET, VIA 	67 	20
li1			PIN 							67 	16
mcon		VIA 							67 	44
mcon		PIN 							67 	41
mcon		NET 							67 	58
met1		NET, SPNET, VIA 	68 	20
met1		PIN 							68 	16
via			VIA 							68 	44
via			PIN 							68 	41
via			NET 							68 	58
met2		NET, SPNET, VIA 	69 	20
met2		PIN 							69 	16
via2		VIA 							69 	44
via2		PIN 							69 	41
via2		NET 							69 	58
met3		NET, SPNET, VIA 	70 	20
met3		PIN 							70 	16
via3		VIA 							70 	44
via3		PIN 							70 	41
via3		NET 							70 	58
met4		NET, SPNET, VIA 	71 	20
met4		PIN 							71 	16
via4		VIA 							71 	44
via4		PIN 							71 	41
via4		NET 							71 	58
met5		NET, SPNET, VIA 	72 	20
met5		PIN 							72 	16
```

mithro, 2021-04-12:
We should totally figure out a process for getting this into the SkyWater PDK....

nguyendao-uom, 2021-04-12:
Thanks @egiacomin, It seems OK for exporting gds file from Innovus, but doesn't show up all drawing layers (only Pin and via layers) when stream in Virtuoso. Do you have a tech file for importing the gds file?

mithro, 2021-04-12:
BTW There is a `#cadence-innovus` channel on the [SkyWater Slack channel](https://join.skywater.tools/) that might be able to help.

egiacomin, 2021-04-13:
@nguyendao-uom I have access to the original PDK with the technology library, so I simply created a new library, attached it to the skywater technology_library and imported my GDS without any layer mapping file. What procedure did you follow?

@mithro should I do a pull request to add the file under the _skywater-pdk/vendor/synopsys/PlaceRoute/common_ directory? Or do you have something else in mind?

nguyendao-uom, 2021-04-13:
@egiacomin : you mean the s8 pdk from skywater? I don't have the access to it.

egiacomin, 2021-04-13:
@nguyendao-uom yes. Have you been able to import any standard cell GDS successfully by any chance? If yes, what mapfile/techfile did you use?

nguyendao-uom, 2021-04-13:
We have TSMC, UMC technologies only. And, of course, we have the NDA for the access.

mattvlsi, 2022-04-13:
Hi @egiacomin. I'm trying to import SOFA FPGA's gds files into Virtuoso or Custom Compiler. By using your map file it successfully imports the gds. And there seems correct layer names and colours. However in layout there are only some vias and small rectangulars. Unable to see standart cells of sky130 and top view of SOFA FPGA's. Can you describe more detailed way how to attach tecnology file and where can I find this technology library? Thanks.

m-usama-z, 2022-10-10:
Hi. I've a question about the layer map file used for exporting GDS from Innovus:
There are some purpose numbers mentioned in the skywater layer info excel sheet, which are not mentioned in the above layer map file, e.g. 13, 10, 60, 0, 14, 25 and 15. Would the absence of these purpose numbers in the map file cause issues in fabrication?

m-usama-z, 2022-10-14:
Kindly add the following line to the map file, whenever it may be incorporated into the PDK:

DIEAREA         ALL       235    4

This is required for density check using Klayout.