google/skywater-pdk #327: Parameter names that shadow device names are not compatible with ngspice

RTimothyEdwards, 2021-05-25:
Some of the parameter names used by the monte carlo device models have the same name as some of the device models themselves.  Apparently ngspice handles parameters by doing a global search-and-replace before parsing contents of a file.  Therefore the device subcircuit itself gets replaced by the parameter contents, which causes immediate failure.

It is not clear whether this is a bug in ngspice or just the way ngspice handles parameters.  However, it can be worked around by changing the name of any parameter that is the same as a device name.  The recommended procedure would be to remove the leading "sky130_fd_pr__" from the parameter name.

Example case:  sky130_fd_pr__nfet_01v8_lvt
Files affected:  The problem seems to be restricted to the two files parameters/critical.spice and parameters/montecarlo.spice.

holvo, 2021-06-12:
There is a fix uploaded to ngspice git branch pre-master.

R and C are not yet covered here, due to the many
instance line options available.

So for now don't use R and C model names and parameters with the same name.

holvo, 2021-06-15:
Tim,

my patch is o.k., but may not be a fix to this problem here.

Could you please provide here a (hopefully simple) example netlist for ngspice which shows the error?

RTimothyEdwards, 2021-06-15:
This issue popped up in the SkyWater PDK.  I am not overly concerned about ngspice's behavior with respect to this, since it's generally bad practice to overlap subcircuit or model names with parameter names, and I have added a patch in open_pdks (eventually to be applied to the sky130_fd_pr repository) to rename the parameters and so avoid the problem altogether.

It was at one time easy to reproduce with the open_pdks install of the sky130A process by running any monte carlo simulation, as ngspice would halt on encountering a subcircuit name in the parameter list from the included file parameters/critical.spice, as mentioned above.  "critical.spice" line 60 has (had, before applying the patch in open_pdks): ".param sky130_fd_pr__nfet_01v8_lvt = 0.0"

I am unable to produce a simple test circuit that reveals this error.  The best I can come up with is the following:
(1) Get open_pdks and configure.  Edit Makefile and remove the two lines that run the script "montecarlo_hack.py" (this script is patching the PDK so that this problem doesn't happen)
(2) Build sky130A from open_pdks (locally, don't need to install).
(3) Run the attached netlist
[test.zip](https://github.com/google/skywater-pdk/files/6656914/test.zip)
(4) The result is "Error: unknown subckt: x1.x0 a b c d x1.0.0+mc_pr_switch*agauss 0 x1.1.0 x1.1 w=1 l=0.5 m=1 mult=1", which suggests that the parameter was expanded in the line that called the subcircuit, overwriting the subcircuit name "sky130_fd_pr__nfet_01v8_lvt" with the definition of the parameter which has the same name as the subcircuit.
(5) As an aside comment, note that it takes ngspice several minutes to start up even though this circuit contains only one transistor, and even with the .spiceinit file setting ng_nomodcheck.  The startup time is proportional to the number of sections defined in the .lib file, even though only one section can ever be active at a time.

holvo, 2021-06-16:
Install with Cygwin: I had to modify file configure a little, because it insisted on requiring the availability of MAGIC.

I have identified the bug: if we have .param pname = xxx, with xxx containing statistical functions (agauss etc.), then ngspice translates them to .func pname xxx, and afterwards does a replacement of all occurrences of pname with this function.

A solution might be to check how many nodes we have in the 
x1 n1 n2 n3 ... pname l=1 w=3
line, and then exclude these plus subckt name from being replaced.

I will have a look where we waste our time a little later.

holvo, 2021-06-18:
I have uploaded anoter fix to ngspice git branch pre-master. Simle example files are now o.k. with the MC portion of the PDK.