google/skywater-pdk #328: Schematic tool for analog design

engrvip, 2021-05-26:
Which tool is supported with sky130 pdk right now for creating analog schematic and running simulation using which SPICE engine ? Can xschem or xcircuit be used right now ?

RTimothyEdwards, 2021-06-14:
xschem is well supported.  For xcircuit, I made some cell libraries but ended up on the problem of how to enforce limits on the parameterized values, which needs some coding work, and I have not gone back to that yet.  Xcircuit can be used, though, just doesn't enforce parameter limits (but I'm not sure xschem does, either).

engrvip, 2021-12-26:
Please share the steps or some reference on how can required schematic editor and simulator can be installed for initiating with analog design using sky130 pdk.