google/skywater-pdk #336: suspicious buf max_capacitance values

maliberty, 2021-09-02:
In examining sky130_fd_sc_hd__tt_025C_1v80.lib I see:

    cell ("sky130_fd_sc_hd__buf_1") {
            max_capacitance : 0.1300150000;
    cell ("sky130_fd_sc_hd__buf_2") {
            max_capacitance : 0.3158670000;
   cell ("sky130_fd_sc_hd__buf_4") {
            max_capacitance : 0.5612280000;
   cell ("sky130_fd_sc_hd__buf_6") {
            max_capacitance : 0.7855320000;
    cell ("sky130_fd_sc_hd__buf_8") {
            max_capacitance : 5.0000000000;
   cell ("sky130_fd_sc_hd__buf_12") {
            max_capacitance : 5.0000000000;
    cell ("sky130_fd_sc_hd__buf_16") {
            max_capacitance : 5.0000000000;

It seems suspicious that the max_capacitance is exactly 5.000 for the 8, 12, and16 sizes of buf.  Is this correct?

RTimothyEdwards, 2021-09-02:
I agree with your suspicions, but the values are from the original SkyWater sources.  The max_transition value for these cells also seems out of line.  There are bufbuf_8/12/16 (double-buffered) and clkbuf_8/12/16 that have values that are more or less what you'd expect extrapolating from buf_4 and buf_6.

maliberty, 2021-09-03:
@RTimothyEdwards how do we ask Skywater about it?