google/skywater-pdk #347: Issue in make timing step

ratul619, 2021-11-28:
Hi , 

I am trying to build the timing libs , but seems some issue : Not able to make out. 
Although using vsdflow , but i tried to do same standalone , But same error : 

chmod a+x /home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/downloads/Miniconda3-latest-Linux-x86_64.sh
/home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/downloads/Miniconda3-latest-Linux-x86_64.sh -p /home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/conda -b -f
touch "/home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/conda/bin/python"
touch: cannot touch '/home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/conda/bin/python': No such file or directory
make: *** [/home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/third_party/make-env/conda.mk:129: /home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/conda/bin/python] Error 1
root@DESKTOP-VLCF3O8:/home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk# ll /home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/conda/bin/python
ls: cannot access '/home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/conda/bin/python': No such file or directory
root@DESKTOP-VLCF3O8:/home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk# ll /home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/conda/bin/
ls: cannot access '/home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/conda/bin/': No such file or directory
root@DESKTOP-VLCF3O8:/home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk# ll /home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/conda/
ls: cannot access '/home/ratul619//vsdflow/work/tools/openlane_working_dir/pdks/skywater-pdk/env/conda/': No such file or directory

Thanks

mithro, 2021-11-28:
I'm afraid you haven't provided enough information for us to help with your problem. Please include the full output of the command you are trying to run.

ratul619, 2021-11-28:
@mithro - I just figured some other issue . I tried to re-run the "make timing" command by turning off my firewall : 

But Still see some issue

root@DESKTOP-VLCF3O8:/home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk# make timing
source /home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk/env/conda/bin/activate && conda env update --name skywater-pdk-scripts --file environment.yml
Collecting package metadata (repodata.json): failed

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://conda.anaconda.org/symbiflow/linux-64/repodata.json>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.
'https://conda.anaconda.org/symbiflow/linux-64'

make: *** [/home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk/third_party/make-env/conda.mk:145: /home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python] Error 1
root@DESKTOP-VLCF3O8:/home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk# make timing
source /home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk/env/conda/bin/activate && conda env update --name skywater-pdk-scripts --file environment.yml
Collecting package metadata (repodata.json): failed

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://conda.anaconda.org/symbiflow/linux-64/repodata.json>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.
'https://conda.anaconda.org/symbiflow/linux-64'

ratul619, 2021-11-28:
I think my env does not have conda installed

mithro, 2021-11-28:
It looks like your internet is broken in some way as the make file was unable to download URLs like https://conda.anaconda.org/symbiflow/linux-64/repodata.json

ratul619, 2021-11-28:
strangely it is able to download and install packages. Right now im checking more, but confused - same as peter parker in mutliverse...

ratul619, 2021-11-28:
I retried again - it did proceee , but hit another wall : 

root@DESKTOP-VLCF3O8:/home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk# make timing
source /home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk/env/conda/bin/activate && conda env update --name skywater-pdk-scripts --file environment.yml
Collecting package metadata (repodata.json): done
Solving environment: done

Downloading and Extracting Packages
tk-8.6.11            | 3.0 MB    | ################################################################################################################################ | 100%
xz-5.2.5             | 341 KB    | ################################################################################################################################ | 100%
_libgcc_mutex-0.1    | 3 KB      | ################################################################################################################################ | 100%
ncurses-6.3          | 782 KB    | ################################################################################################################################ | 100%
ca-certificates-2021 | 115 KB    | ################################################################################################################################ | 100%
pip-21.2.4           | 1.8 MB    | ################################################################################################################################ | 100%
zlib-1.2.11          | 103 KB    | ################################################################################################################################ | 100%
setuptools-58.0.4    | 790 KB    | ################################################################################################################################ | 100%
sqlite-3.36.0        | 990 KB    | ################################################################################################################################ | 100%
libffi-3.3           | 50 KB     | ################################################################################################################################ | 100%
libgcc-ng-9.3.0      | 4.8 MB    | ################################################################################################################################ | 100%
python-3.8.12        | 18.3 MB   | ################################################################################################################################ | 100%
_openmp_mutex-4.5    | 22 KB     | ################################################################################################################################ | 100%
ld_impl_linux-64-2.3 | 586 KB    | ################################################################################################################################ | 100%
libstdcxx-ng-9.3.0   | 3.1 MB    | ################################################################################################################################ | 100%
libgomp-9.3.0        | 311 KB    | ################################################################################################################################ | 100%
openssl-1.1.1l       | 2.5 MB    | ################################################################################################################################ | 100%
certifi-2021.10.8    | 151 KB    | ################################################################################################################################ | 100%
readline-8.1         | 362 KB    | ################################################################################################################################ | 100%
wheel-0.37.0         | 33 KB     | ################################################################################################################################ | 100%
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
Installing pip dependencies: \ Ran pip subprocess with arguments:
['/home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk/condaenv.l6jo42z2.requirements.txt']
Pip subprocess output:
Obtaining file:///home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk/scripts/python-skywater-pdk (from -r /home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk/requirements.txt (line 8))

Pip subprocess error:
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f206a9d37f0>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/flake8/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f206a9d33d0>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/flake8/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f206a9d3790>: Failed to establish a new connection: [Errno -2]  [... cut ...]
/home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python: Error while finding module specification for 'skywater_pdk.liberty' (ModuleNotFoundError: No module named 'skywater_pdk')
/home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python: Error while finding module specification for 'skywater_pdk.liberty' (ModuleNotFoundError: No module named 'skywater_pdk')

mithro, 2021-11-28:
Again these look like network issues....

ratul619, 2021-11-29:
Some isse in my machine. Either connection/system issue. 
Did try building in another machine and it worked.
Thanks

closed 2021-11-29 by ratul619

chchzh1978, 2024-07-12:
> I retried again - it did proceee , but hit another wall :
>
> root@DESKTOP-VLCF3O8:/home/ratul619//vsdflow/work/tools/openlane_working_dir/openlane/pdks/skywater-pdk# make timing source [...]

This happen because the network is not stable when doing "make timing", in order to solve this issue, you can remove the ./env,  then do "make timing" again.