google/skywater-pdk #348: IO filler cell name ?

ratul619, 2021-12-08:
Hi , 

Which cells is/are the IO filler cells? Not able to find in documentation , neither any candidate seems suitable in LEF . 

Thanks

RTimothyEdwards, 2021-12-08:
Apparently you are not using the open_pdks installer [https://github.com/RTimothyEdwards/open_pdks] which is where the I/O filler cells are?

ratul619, 2021-12-08:
I think i directly installed from the skywater-pdk git repo. I will look into the open_pdks . Thanks

RTimothyEdwards, 2021-12-08:
Look in `sky130_ef_io.lef`. The fill cells are:

sky130_ef_io__com_bus_slice_1um
sky130_ef_io__com_bus_slice_5um
sky130_ef_io__com_bus_slice_10um
sky130_ef_io__com_bus_slice_20um

ratul619, 2021-12-08:
Ah. Makes sense. I will have a look at those . Thanks @RTimothyEdwards

closed 2021-12-08 by ratul619