google/skywater-pdk #349: Wrong SPICE netlist of sky130_fd_sc_hvl__lsbuflv2hv_1

hpretl, 2021-12-17:
## Expected Behavior

LVS should match for SPICE netlist and GDS.

## Actual Behavior

It ain't.

## Steps to Reproduce the Problem

First extract netlist with `magic`.

`netgen -batch lvs "sky130_fd_sc_hvl__lsbuflv2hv_1.spice sky130_fd_sc_hvl__lsbuflv2hv_1" "lvs_cell.spice sky130_fd_sc_hvl__lsbuflv2hv_1" $PDKPATH/libs.tech/netgen/sky130A_setup.tcl`

## Specifications

- Version: latest
- Platform: n/a

RTimothyEdwards, 2021-12-17:
It's not clear to me which netlist is being compared against which netlist.  Can you please give a clear indication of where each of the netlists came from?

hpretl, 2021-12-17:
Sorry for not being clear enough. It is this SPICE netlist

[sky130_fd_sc_hvl__lsbuflv2hv_1.spice](https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hvl/blob/4fd4f858d16c558a6a488b200649e909bb4dd800/cells/lsbuflv2hv/sky130_fd_sc_hvl__lsbuflv2hv_1.spice)

compared against the extracted netlist from this GDS

[sky130_fd_sc_hvl__lsbuflv2hv_1.gds](https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hvl/blob/4fd4f858d16c558a6a488b200649e909bb4dd800/cells/lsbuflv2hv/sky130_fd_sc_hvl__lsbuflv2hv_1.gds)

(both from the current `sky130_fd_sc_hvl` library. There might be more errors in the SPICE netlists, this is the first cell I checked.

RTimothyEdwards, 2021-12-17:
There are issues due to the fact that the SkyWater netlist uses the same names for power and ground nodes that are separate rails and therefore electrically disconnected.  I'm not sure, though, why extracting without "extract unique" won't give a matching netlist.  I do know that extracting with "extract unique" will give a topological match but not a pin match.  What version of magic and netgen did you use?

RTimothyEdwards, 2021-12-17:
Okay, on inspection of the .spice file in the skywater-pdk repository, it somehow dissassociated the two redundant net names from their respective pins and turned them into internal nodes.  The immediate solution is to not trust the SPICE from skywater-pdk and generate netlists from GDS during the open_pdks install.  As a side note, this problem can only show up on a handful of cells such as the level shifters in the HVL library. I don't think any of the other libraries have any double-height standard cells.

hpretl, 2021-12-18:
I used the most recent versions from GitHub:

`magic` 5e3c26c95afadfd07753bb8126fa6528afce5dd8 tag: 8.3.241
`netgen` de18ae85a4f0a96271a301a4704d42e7ac132963 from Dec/15

Your analysis sounds correct and fits the error. My guess is that the SPICE netlists have been created/extracted at a time when this "join nets by name" was not properly working in `magic` extract. With the current versions of `magic` and `netgen` it works properly, which is good.

That leaves a check and fix of the sky130 cells as to be done.

d-m-bailey, 2021-12-24:
Similar to #305.