google/skywater-pdk #351: Off-grid standard cells

d-m-bailey, 2021-12-24:
## Expected Behavior
All layout patterns should be on grid.

## Actual Behavior
When reading gds into magic, offgrid warnings are generated.
```
Reading "sky130_fd_sc_hd__a2111o_1".
CIF file read warning: CIF style sky130(vendor): units rescaled by factor of 5 / 1
CIF file read warning: Input off lambda grid by 2/5; snapped to grid.
```

## Steps to Reproduce the Problem

1. `cd $OPENLANE_ROOT`
2. `git clone https://github.com/efabless/caravel.git`
1. `make mount`
1. `cd caravel/openlane`
2.  `gunzip ../gds/housekeeping.gds.gz`
2. `flow.tcl -design housekeeping -lvs -gds ../gds/housekeeping.gds -net ../verilog/gl/housekeeping.v` 

## Specifications

- Version:  `skywater-pdk c094b6e83a4f9298e47f696ec5a7fd53535ec5eb`
`open_pdks 476f7428f7f686de51a5164c702629a9b9f2da46`
- Platform: `Linux ciic-cvc 4.19.0-17-cloud-amd64 #1 SMP Debian 4.19.194-3 (2021-07-18) x86_64 GNU/Linux`

RTimothyEdwards, 2021-12-24:
@d-m-bailey You need to be specific---is this the skywater PDK or open_pdks?  Because the SkyWater a2111o_1 cell has a DRC error (undetected by Calibre---never understood why) which I fixed, so open_pdks has a wholesale replacement of the GDS for the a2111o_1 cell.  If it's the open_pdks that's the issue, then the issue is my fault and it should be raised in the open_pdks repository.  Possibly I created an off-grid point when I did that.  I would have to take a look at the GDS I created.

RTimothyEdwards, 2021-12-24:
@d-m-bailey :  I realize that's not an easily answered question.  I'm trying to figure it out now.

RTimothyEdwards, 2021-12-24:
@d-m-bailey :  Aaaaand. . . it's not a2111o_1 that was customized, it was a2111oi_0.  So I retract all those statements above and will now go staring at the SkyWater cell to see what's up with it.

RTimothyEdwards, 2021-12-24:
```
    0x0006     # RECORD_LENGTH              Bytes of data in this record
    0x0d       # RECORD_TYPE:  LAYER        Layer specification
    0x02       # DATA_TYPE:    INTEGER_2    Two byte signed integer
    0x007a     # DATA: 122

    0x0006     # RECORD_LENGTH              Bytes of data in this record
    0x0e       # RECORD_TYPE:  DATATYPE     Datatype specification
    0x02       # DATA_TYPE:    INTEGER_2    Two byte signed integer
    0x0010     # DATA: 16

    0x002c     # RECORD_LENGTH              Bytes of data in this record
    0x10       # RECORD_TYPE:  XY           An array of XY coordinates
    0x03       # DATA_TYPE:    INTEGER_4    Four byte signed integer
    0x000007b7 # DATA: 1975
    0xffffffbd # DATA: -67
    0x00000861 # DATA: 2145
    0xffffffbd # DATA: -67
    0x00000861 # DATA: 2145
    0x00000067 # DATA: 103
    0x000007b7 # DATA: 1975
    0x00000067 # DATA: 103
    0x000007b7 # DATA: 1975
    0xffffffbd # DATA: -67
```

Sure enough, there's a shape on layer 122:16 (pwell pin) that's completely off-grid.  I will prepare another fixed version in open PDKs.  Fortunately, it's a pin label layer, not a mask layer, so it won't show up as a manufacturing error.

RTimothyEdwards, 2021-12-24:
@d-m-bailey :  open_pdks has now been updated with a custom corrected cell layout GDS file.