google/skywater-pdk #354: QRC tech file

WahidSadiq, 2022-01-14:
I am trying to implement this PDK in genus and innovus. For mmmc we need captable or qrc tech file. But i didn't find it. is there any way to get it?

QuantamHD, 2022-01-14:
This PDK is mostly targeted at the opensource eda tools, and we have limited support for commercial EDA tools.

It should be possible to create those files from the existing data in this repo. We would happily accept pull requests to enhance the PDK in that direction.

@RTimothyEdwards do you have any insight on these specific files?

WahidSadiq, 2022-01-14:
thanks man. it will be very helpful to get the data related to QRC tech file so that we can create it

rovinski, 2022-01-14:
Maybe take a look at #185 and #187. That might help get you started.

WahidSadiq, 2022-01-14:
> Maybe take a look at #185 and #187. That might help get you started.

thanks @rovinski i think this is enough to get myself started

closed 2022-01-14 by WahidSadiq