google/skywater-pdk #357: Read-the-docs is missing sky130_fd_sc_lp

maliberty, 2022-02-05:
## Expected Behavior
https://antmicro-skywater-pdk-docs.readthedocs.io/en/test-submodules-in-rtd/contents/libraries/foundry-provided.html
has pages for each stdcell variant but sky130_fd_sc_lp under PDK Contents/Libraries.

## Actual Behavior
sky130_fd_sc_lp has a page like https://antmicro-skywater-pdk-docs.readthedocs.io/en/test-submodules-in-rtd/contents/libraries/sky130_fd_sc_hd/README.html

## Steps to Reproduce the Problem

1. Browse to link above

mithro, 2022-05-04:
FYI - @umarcor started looking at fixing a lot of the outstanding documentation issues with the SkyWater PDK.

proppy, 2022-07-01:
related to https://github.com/google/skywater-pdk/pull/143, since we should also update [skywater-pdk.rtfd.io](https://skywater-pdk.rtfd.io/) to have those index in the first place.