google/skywater-pdk #359: Please correct resistor values in the PDK documentation

RTimothyEdwards, 2022-02-09:
The documentation has added values for contact resistances taken from my spreadsheet.  However, upon reviewing some of the SkyWater etest data, I realize that I had written down the value for the LICON-to-poly contact resistance under the MCON contact.

The document page needing correcting is here:  https://skywater-pdk.readthedocs.io/en/main/rules/rcx.html#resistance-rules

## Existing Documentation

LICON contact | 15000
MCON contact | 152000

## Correct Documentation

LICON contact to poly | 152000
LICON contact to ndiff | 185000
LICON contact to pdiff | 585000
MCON contact | 9300

Note that there are significant differences in the LICON resistance depending on what material it contacts, so there should be three separate entries for this contact resistance.

RTimothyEdwards, 2022-02-09:
FYI, I have corrected the spreadsheet (https://docs.google.com/spreadsheets/d/1oL6ldkQdLu-4FEQE0lX6BcgbqzYfNnd1XA8vERe0vpE/edit#gid=1059412464)

proppy, 2022-07-01:
Looks like those are sourced from a CSV file included in https://github.com/google/skywater-pdk/pull/106.