google/skywater-pdk #361: Calibre LVS

mousaq92, 2022-03-01:
Hello,

I apologize for not following the issue template. When will the Calibre LVS decks be available?

Thanks,

Mousa

proppy, 2022-07-01:
Looking at https://code.stanford.edu/jjthomas/skywater-130nm-adk README, it seems that the project has plans to generate calibre lvs deck:
```
calibre-drc-block.rule
calibre.layerprops
calibre-lvs.rule
```
> The final three calibre files are not available yet. We need to create some scripts that generate these files from the technology information in the PDK.

Maybe @priyanka-raina can comment on the status?

priyanka-raina, 2022-07-03:
It is non trivial to generate the Calibre decks, and we do not have the expertise to do so. If you would like access to Calibre decks, the best way is to sign an NDA with SkyWater and get access to these files directly from them.

mithro, 2022-07-06:
At the moment we don't have any short term plans to produce Calibre LVS decks in this PDK. You should instead be able to use Magic or KLayout for DRC and LVS checking.

closed 2022-07-06 by mithro