google/skywater-pdk #365: Question about how skywater-pdk LEF file is used by OpenLane

jarvis-lorry, 2022-04-08:
In the sky130A LEF file, for the cell macros, I can see interconnect (li1) layers for pins and power/ground, but the other layers and connections in the final layout aren’t included there. For example, the li1 drawing layer and diffusion layers are added later. This may seem an ignorant question, but can anyone tell me how OpenLane gets the extra information needed to draw in the cells?

RTimothyEdwards, 2022-04-09:
@jarvis-lorry :  Openlane needs to read the **technology LEF** file in addition to the standard cell macros.  In the open_pdks installation, you'll find, for example, the HD library technology LEF file in `libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd__nom.tlef`.

jarvis-lorry, 2022-04-11:
Thank you, Timothy.  I can see that the technology LEF file has the rules for width, pitch and offset which are used by OpenLane to draw out the circuits.

jarvis-lorry, 2022-05-24:
@RTimothyEdwards I had another question--maybe a dumb one--about the skywater tech LEF.  Are the resistance, capacitance and edge capacitance values in the techlef used in OpenLane place and route or are they used only for RC modeling for back annotation in timing?

proppy, 2022-07-01:
@jarvis-lorry that may be a better requestion for the https://github.com/The-OpenROAD-Project/OpenROAD (or https://github.com/The-OpenROAD-Project/OpenLane) as they are the project consuming those files.