google/skywater-pdk #370: Problems with downloading the 'latest' version of standard cell libraries

zorrolee777, 2022-05-16:
## Expected Behavior
When cloning the repo and running make submodules it should pull all the submodules.

## Actual Behavior
Got some errors when running the following commands.

At beginning I received "**SUBMODULE_VERSION=latest: Command not found.fatal: unable to write new index file**", but it can keep going, and eventually received another massage: 
**Unable to checkout 'ae1b7f68821505cf2d93d9d44cce5ece22710fad' in submodule path 'libraries/sky130_fd_sc_ms/latest'
make: *** [libraries/sky130_fd_sc_ms/latest/.git] Error 1**

## Steps to Reproduce the Problem

1. `git clone https://github.com/google/skywater-pdk`
2. `SUBMODULE_VERSION=latest make submodules -j3 || make submodules -j1`

## Specifications

- Version:
- Platform:

mithro, 2022-05-16:
I'm unable to reproduce this issue - Can you please include the full log somewhere?

proppy, 2022-07-01:
Also enable to reproduce, feel free to reopen by commenting on this issue after attaching the full log.

closed 2022-07-01 by proppy