google/skywater-pdk #371: make timing fails in conda set up, seems flaky

growly, 2022-05-25:
## Expected Behavior
`make timing` to conclude

## Actual Behavior
`make timing` errors out during a conda step. Looks like a `condaenv.[something].requirements.txt` file is missing.

[make timing output](https://github.com/google/skywater-pdk/files/8774498/make_timing.log)
[bash environment variables](https://github.com/google/skywater-pdk/files/8774504/env.txt)

I've actually made it past this point a couple of times, and I can't figure out what's different between the invocations.

## Steps to Reproduce the Problem

I suspect this is related to my environment, but I'm not sure what to check first.

1. git clone this repo; cd there
1. git checkout  00bdbcf4a3aa922cc1f4a0d0cd8b80dbd73149d3
1. SUBMODULE_VERSION=latest make submodules -j3 || make submodules -j1
2. make timing

## Specifications

- Version: 00bdbcf4a3aa922cc1f4a0d0cd8b80dbd73149d3
- Platform: 00bdbcf4a3aa922cc1f4a0d0cd8b80dbd73149d3

mithro, 2022-05-25:
Could it be an issue caused by the proxy configuration? `AUTO_PROXY=https://proxyconfig.corp.google.com/proxy.pac`?

mithro, 2022-05-25:
The lack of output from `pip` makes it very hard to see what is going on....

mithro, 2022-05-25:
I'm getting this error;
```
Ran pip subprocess with arguments:
['/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python', '-m', 'pip', 'install', '-U', '-r', '/ssd/github/google/skywater-pdk/condaenv.ggibcadt.requirements.txt']
Pip subprocess output:

Pip subprocess error:
ERROR: Exception:
Traceback (most recent call last):
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", line 173, in _main
    status = self.run(options, args)
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/cli/req_command.py", line 203, in wrapper
    return func(self, options, args)
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/commands/install.py", line 286, in run
    reqs = self.get_requirements(args, options, finder, session)
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/cli/req_command.py", line 384, in get_requirements
    for parsed_req in parse_requirements(
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 135, in parse_requirements
    for parsed_line in parser.parse(filename, constraint):
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 319, in parse
    yield from self._parse_and_recurse(filename, constraint)
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 348, in _parse_and_recurse
    yield from self._parse_and_recurse(req_path, nested_constraint)
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 324, in _parse_and_recurse
    for line in self._parse_file(filename, constraint):
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 353, in _parse_file
    _, content = get_file_content(filename, self._session)
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/req/req_file.py", line 518, in get_file_content
    resp = session.get(url)
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_vendor/requests/sessions.py", line 555, in get
    return self.request('GET', url, **kwargs)
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_internal/network/session.py", line 454, in request
    return super().request(method, url, *args, **kwargs)
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_vendor/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_vendor/requests/sessions.py", line 697, in send
    r.content
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_vendor/requests/models.py", line 836, in content
    self._content = b''.join(self.iter_content(CONTENT_CHUNK_SIZE)) or b''
  File "/ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/site-packages/pip/_vendor/requests/models.py", line 769, in generate
    chunk = self.raw.read(chunk_size)
AttributeError: 'FileNotFoundError' object has no attribute 'read'

CondaEnvException: Pip failed

make: *** [/ssd/github/google/skywater-pdk/third_party/make-env/conda.mk:145: /ssd/github/google/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python] Error 1
```

mithro, 2022-05-25:
@growly - Why are you checking out 00bdbcf4a3aa922cc1f4a0d0cd8b80dbd73149d3 ? The commit right cc1db3eb708f74a9f2c98cec9bb6d8cd4e48af30 after fixes the issue....

growly, 2022-05-25:
That's the commit openlane specifies [here](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/manual_pdk_installation.md). No doubt it's out of date.

Thanks. Weird that it works sometimes?

closed 2022-05-25 by growly

mithro, 2022-05-25:
Can you log a bug where ever you found that commit hash?