google/skywater-pdk #373: MOSFET Simulation Parameters

SpaceDreams, 2022-06-02:
Thanks for offering something like this! I'm trying to get started with some ngspice simulations for the mosfets in the sky130_pr library. I haven't looked at any layouts or DRC's so I'm wondering what would be appropriate values for: as ad ps  pd (in terms of w, l and NF). Thanks!

nataraj-pinakapani, 2022-06-06:
ad='int((nf+1)/2) * W/nf * 0.29' 
as='int((nf+2)/2) * W/nf * 0.29'
pd='2*int((nf+1)/2) * (W/nf + 0.29)' 
ps='2*int((nf+2)/2) * (W/nf + 0.29)' 
nrd='0.29 / W' 
nrs='0.29 / W'

SpaceDreams, 2022-06-06:
Thanks alot! Do you have any references for this? (I mean, did you use the DRC to get the minimum dimensions for the source and drain?) Lastly, is the 0.29 in micro-meters? Again, thanks so much!