google/skywater-pdk #374: ngspice-36-klu for simulation with google/skywater-pdk

holvo, 2022-06-21:
Users have reported some problems when simulating large circuits (e.g. 200k transistors) with standard ngspice (long simulation times, operating point convergence problems).

Recently I have tested a dedicated version of ngspice-36, which replaces the venerable Sparse matrix solver by the KLU method, with very promising results. When using KLU and the Optran method to get the operating point before transient sim is started, I have achieved a simulation speed-up by a factor of three or more, after a fast operating point evaluation.

On the ngspice web pages at http://ngspice.sourceforge.net/applic.html there is more information on setting up ngspice for KLU. 

As user feedback is very important here, please don't hesitate to try it out and report your results. We will need much more testing with various circuits before this feature may move into the standard ngspice distribution.

proppy, 2022-07-01:
@holvo pinned the issue for more visibility.

proppy, 2022-07-01:
@holvo  what the best venue for PDK users to give feedback to the ngspice community?

holvo, 2022-07-01:
ngspice developers are active in the ngspice discussion forums, especially [ngspice-devel](https://sourceforge.net/p/ngspice/discussion/127605/) . ngspice bugs may be reported to the bug tracker at https://sourceforge.net/p/ngspice/bugs/.

holvo, 2024-01-26:
Meanwhile ngspice-42 is available, containing the KLU solver, when compiled with --enable-klu, so I will close this issue.

closed 2024-01-26 by holvo