google/skywater-pdk #375: spice simulations not found

joamatab, 2022-06-21:
Hi!

im trying to load the spice simulations into pyspice

When running `sky130/spice_models.py`

```
https://github.com/gdsfactory/skywater130/blob/main/sky130/spice_models.py
```

i get

```
ERROR, library file /home/jmatres/sky130/sky130/spice/sky130_fd_sc_hd.spice, section definition sky130_fd_sc_hd__a2111o_1 not f
ound
```

what is the format for loading spice simulations in pyspice?

https://github.com/gdsfactory/skywater130/issues/16

mithro, 2022-06-21:
If you install the PDK using [open_pdks](https://github.com/RTimothyEdwards/open_pdks) you should get something which is usable with pyspice

proppy, 2022-07-01:
Yes, I'd recommend to install a packaged version of the PDK thru https://anaconda.org/LiteX-Hub/open_pdks.sky130a or https://github.com/efabless/volare.

In particular you want to make sure you load the following file with `.lib`: https://github.com/RTimothyEdwards/open_pdks/blob/master/sky130/custom/models/sky130.lib.spice

closed 2022-07-01 by proppy

proppy, 2022-07-01:
@RTimothyEdwards would it make sense to include https://github.com/RTimothyEdwards/open_pdks/tree/master/sky130/custom/models in this repo instead? So that project like gdsfactory can more easily repackage/redistribute the spice models?

RTimothyEdwards, 2022-07-01:
@proppy :  There is this and about a hundred other fixes and enhancements to the PDK that would be better to have in the PDK itself rather than in open_pdks.  open_pdks has become the only place to put fixes for skywater-pdk because skywater-pdk never gets updates.  I would much prever that open_pdks has little more to do than to collate some individual files into libraries.  So YES, it makes sense to apply all those updates, patches, fixes, and enhancements to the skywater-pdk.  But issuing pull requests on skywater-pdk has never accomplished anything in the past.  If that has changed, let me know.

proppy, 2022-07-06:
> But issuing pull requests on skywater-pdk has never accomplished anything in the past. If that has changed, let me know.

Yes, I want to actively work on reconciling those changes with this repo (ideally https://github.com/RTimothyEdwards/open_pdks/tree/master/sky130/custom/ would be no more), I understand this is no small work but I think it's critical so that we can enable smoother collaboration between the PDK repo and other tools.