google/skywater-pdk #376: tap cells LEF 5.5 - CLASS CORE

rachanaerra, 2022-06-29:
## Expected Behavior
Tap cells in LEF version 5.5 should have CLASS CORE. 

## Actual Behavior
The lef (version 5.5) shows CLASS CORE WELLTAP.

[](https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hs/blob/1d051f49bfe4e2fe9108d702a8bc2e9c081005a4/cells/tap/sky130_fd_sc_hs__tap_1.lef)
The link shows lef 5.5 for tap cells.

WELLTAP statement is for version 5.6 and later.

## Steps to Reproduce the Problem
1.
1.
1.

## Specifications

- Version:
- Platform:

proppy, 2022-07-01:
Looks like this was done in https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/commit/e95890ac25f7109e8cb0347b586a41edb48f52f3 as part of #124 to address #113.

@mithro @msaligane any reason we didn't bump the version in https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/blob/ac7fb61f06e6470b94e8afdf7c25268f62fbd7b1/cells/tap/sky130_fd_sc_hd__tap_1.lef#L17 as part of the same change?

msaligane, 2022-07-01:
@proppy I am actually not sure either.