google/skywater-pdk #380: Release of continuous SPICE MOSFET models

hpretl, 2022-07-05:
## Issue

Currently, the only available SPICE transistor models are binned models. The model binning causes huge model files (leading to a slow simulation startup) and discontinuities across parameter sweep, especially *W* and *L*. 

The release of continuous SPICE MOSFET models would solve all of these issues.

QuantamHD, 2022-07-05:
I don't believe the continuous models exist. I could be totally wrong, but in the PDK provided to us by SkyWater in both the new and old version the models are binned.

We're currently working with some third parties to recharaterize the models, but that work is still ongoing. Feel free to correct me if I'm off though.

@RTimothyEdwards anything you can add?

hpretl, 2022-07-05:
These continuous models apparently do exist, but are not yet released. Even if binning is used, usually there are only a few bins to break down the huge *W* and *L* variations into something more manageable, but this is not done as fine-grained as in the currently released SkyWater PDK.

hpretl, 2022-07-05:
I have no access to the NDA-versions of the PDKs, so I can't check myself. I was just believing that they should be available in SkyWater, at least that is my (and many analog designers) hope.

QuantamHD, 2022-07-05:
I've confirmed that they do in fact exist. The blocker seems to be the conversion from the proprietary formats is not in state that easily allows us to review it. I'll see if we can rectify it.

mattvenn, 2022-07-27:
any progress on this? I know a lot of people are asking for it.

RTimothyEdwards, 2022-07-27:
@proppy has agreed to help with this, so as soon as I am able to get to it, we will get this merged into the repository.

RTimothyEdwards, 2022-07-27:
Since it has not been mentioned in this issue, I should add that I have the converted and working continuous models in a form compatible with ngspice, integrated into the PDK, and generated a patch file to apply to the repository, which was completed maybe four months ago or more.  It only requires someone on the Google side to merge it.

QuantamHD, 2022-07-27:
There is a mild complicating factor. The patch as currently formatted doesn't allow an easy comparison to the base of this repo since there was a reorganization step applied to the directory structure. This means that it requires another line by line manual review of the files in question.

@mithro has told me that we need to generate a patch against the current directory structure before he is willing to review.

QuantamHD, 2022-07-27:
@RTimothyEdwards if we can generate another patch that follows that restriction it should allow an easy export process. I have looked into generating one, and @proppy has done the same. We haven't made a lot of progress since it's always a bit unclear which tools needs to be run from what branch to generate the correct output.

RTimothyEdwards, 2022-07-27:
@QuantamHD :  I have all the scripts that I used to generate the files in the patch from the base s130 repository.  However, those scripts deal with the private SkyWater repository we have under NDA, so I cannot make them public.  But I am happy to hand off everything I have to you and @proppy and then we can work through it properly.

QuantamHD, 2022-07-27:
That sounds like a plan. If you can push those scripts to the google source repo we can start modifying it to generate a diff that will allow a quick review and release.

RTimothyEdwards, 2022-07-27:
@QuantamHD :  Which repository, and which branch?

QuantamHD, 2022-07-27:
https://foss-eda-tools.googlesource.com/skywater-pdk-audit/ 

new branch, you choose the name

RTimothyEdwards, 2022-07-27:
@QuantamHD :  Got it;  will do.

RTimothyEdwards, 2022-07-27:
@QuantamHD :  Pushed as branch `continuous_models`.  I did not attempt to integrate this into skywater-pdk-audit itself;  I just dumped what I have into a new directory `s130_models`.  There is a README file in that directory that has instructions on how to build the ngspice-compatible models and run the device tests.

Also please note that these scripts _only_ generate the ngspice-compatible continuous models and associated files.  The patch file I generated is the result of a lot more work which was largely done by hand, so sort through the existing and new PDKs and determine what needs to be kept, what needs to be replaced, and creating a new sky130.lib.spice that has the "include" statements to point to the correct files.  In that patch, I did not attempt to remove the existing discrete models, so the continuous models created by the scripts I just put in pdk-audit are just put into a directory called `continuous`.  But you can use the scripts in pdk-audit to check how the new model files were built, and to have the scripts available to be able to produce the same from the SkyWater sources.

proppy, 2022-07-28:
@RTimothyEdwards Thanks for pushing this!

I gave you script a try and was able to get up until the last command, where I encounter the following issue:
```
+ /home/proppy/src/github.com/RTimothyEdwards/open_pdks/sky130/custom/scripts/process_params.py ngspice
Traceback (most recent call last):
  File "/home/proppy/src/github.com/RTimothyEdwards/open_pdks/sky130/custom/scripts/process_params.py", line 90, in <module>
    std_dev = float(gparam[1])
ValueError: could not convert string to float: '0.375e-10*process_mc_factor'
```

It seems that https://github.com/RTimothyEdwards/open_pdks/blob/8fe7f760ece2bb49b1c310e60243f0558977dae5/sky130/custom/scripts/process_params.py#L90 assumes a that `std` parameters are always float while the converted models are actual expressions:
```
find . -type f -exec grep --color=auto -nH --null -e process_mc_factor \{\} +
./models_global.spice\0125:*          vary sw_tox_lv_mc             dist=gauss   std=0.375e-10*process_mc_factor
./models_global.spice\0126:*          vary sw_tox_hv_mc             dist=gauss   std=1.25e-10*process_mc_factor
./models_global.spice\0127:*          vary sw_polycd                dist=gauss   std=1.5e-9*process_mc_factor
./models_global.spice\0128:*          vary sw_activecd              dist=gauss   std=5.0e-9*process_mc_factor
./models_global.spice\0129:*          vary sw_fox_cap               dist=gauss   std=0.0475*process_mc_factor
./models_global.spice\0130:*          vary sw_li_cap                dist=lnorm   std=0.1*process_mc_factor
./models_global.spice\0131:*          vary sw_fox_poly_cap          dist=gauss   std=0.0475*process_mc_factor
./models_global.spice\0132:*          vary sw_nw_rs_mult            dist=lnorm   std=0.12*process_mc_factor
./models_global.spice\0133:*          vary sw_pw_rs_mc              dist=gauss   std=6.6*process_mc_factor       percent=yes
./models_global.spice\0134:*          vary sw_nldd                  dist=gauss   std=2.5*process_mc_factor       percent=yes
./models_global.spice\0135:*          vary sw_nsd_pw_cj             dist=gauss   std=2.5*process_mc_factor       percent=yes
./models_global.spice\0136:*          vary sw_psd_nw_cj             dist=gauss   std=2.5*process_mc_factor       percent=yes
./models_global.spice\0137:*          vary sw_nw_pw_cj              dist=gauss   std=2.5*process_mc_factor       percent=yes
./models_global.spice\0138:*          vary sw_pw_dnw_cj             dist=gauss   std=2.5*process_mc_factor       percent=yes
./models_global.spice\0139:*          vary sw_dnw_sub_cj            dist=gauss   std=2.5*process_mc_factor       percent=yes
./models_global.spice\0140:*          vary sw_vth0_sky130_fd_pr__nfet_01v8_mc          dist=gauss   std=0.006*process_mc_factor
./models_global.spice\0141:*          vary sw_vth0_sky130_fd_pr__nfet_01v8_lvt_mc      dist=gauss   std=0.006*process_mc_factor
./models_global.spice\0142:*          vary sw_vth0_sky130_fd_pr__nfet_g5v0d10v5_mc       dist=gauss   std=0.010*process_mc_factor
./models_global.spice\0143:*          vary sw_vth0_sky130_fd_pr__nfet_01v8_nat_mc      dist=gauss   std=0.006*process_mc_factor
./models_global.spice\0144:*          vary sw_vth0_sky130_fd_pr__nfet_g5v0d16v0_mc   dist=gauss   std=0.035*process_mc_factor
./models_global.spice\0145:*          vary sw_vth0_sky130_fd_pr__nfet_01v8_zvt         dist=gauss   std=0.007*process_mc_factor
./models_global.spice\0146:*          vary sw_vth0_sky130_fd_pr__pfet_01v8_mc          dist=gauss   std=0.009*process_mc_factor
./models_global.spice\0147:*          vary sw_vth0_sky130_fd_pr__pfet_01v8_lvt_mc      dist=gauss   std=0.023*process_mc_factor
./models_global.spice\0148:*          vary sw_vth0_sky130_fd_pr__pfet_01v8_hvt_mc      dist=gauss   std=0.009*process_mc_factor
./models_global.spice\0149:*          vary sw_vth0_sky130_fd_pr__pfet_g5v0d10v5_mc       dist=gauss   std=0.02*process_mc_factor
./models_global.spice\0150:*          vary sw_vth0_sky130_fd_pr__pfet_g5v0d16v0_mc   dist=gauss   std=0.03*process_mc_factor
./models_global.spice\0151:*          vary sw_cap_mim_ca            dist=gauss   std=2.7*process_mc_factor       percent=yes
./models_global.spice\0152:*          vary sw_cap_mim_dw            dist=gauss   std=24e-9*process_mc_factor
./models_global.spice\0153:*          vary sw_sky130_fd_pr__res_generic_m1_rs                dist=gauss   std=5*process_mc_factor         percent=yes
./models_global.spice\0154:*          vary sw_sky130_fd_pr__res_generic_m2_rs                dist=gauss   std=5*process_mc_factor         percent=yes
./models_global.spice\0155:*          vary sw_sky130_fd_pr__res_generic_m3_rs                dist=gauss   std=5*process_mc_factor         percent=yes
./models_global.spice\0156:*          vary sw_sky130_fd_pr__res_generic_m4_rs                dist=gauss   std=5*process_mc_factor         percent=yes
./models_global.spice\0157:*          vary sw_m3_dw                 dist=gauss   std=1.6e-8*process_mc_factor
./models_global.spice\0158:*          vary sw_m4_dw                 dist=gauss   std=1.6e-8*process_mc_factor
./models_global.spice\0159:*          vary sw_rcvia                 dist=lnorm   std=0.26*process_mc_factor
./models_global.spice\0160:*          vary sw_rcvia2                dist=lnorm   std=0.21*process_mc_factor
./models_global.spice\0161:*          vary sw_rcvia3                dist=lnorm   std=0.21*process_mc_factor
[... cut ...]
./models_global.spice\0171:*          vary sw_poly_head_res         dist=gauss   std=2.5*process_mc_factor       percent=yes 
./models_global.spice\0172:*          vary sw_sky130_fd_pr__res_high_po_rs           dist=gauss   std=3.5*process_mc_factor       percent=yes
./models_global.spice\0173:*          vary sw_sky130_fd_pr__res_xhigh_po_rs         dist=gauss   std=4.0*process_mc_factor       percent=yes     
```

Is that something we should fix in the conversion process to get compute the scalar value and get rid of the express, or should we make `process_params.py` able to parse (and evaluate?) those?

proppy, 2022-07-28:
@mithro and @QuantamHD as a lot of people are waiting for this, what do you think of dumping the converted models into a separate branch/repo and working together with the community to integrate those properly in the PDK?

RTimothyEdwards, 2022-07-28:
@proppy :  I'm not sure what happened with the process_params.py script.  The README file states that I needed to make some minor changes to the mismatch_params.py and process_params.py files for the different syntax.  The modified mismatch_params.py got copied back into the local directory but the process_params.py didn't.  My best guess is that I originally corrected the routines in place, and then forgot that I had modified process_params.py and didn't copy it back into the directory locally.  I have since overwritten my change, but I know what it was, so I will fix it and push the correction.

RTimothyEdwards, 2022-07-28:
@proppy :  I confirmed that I had modified the process_params.py script only in a local branch I made of open_pdks, so using the distribution open_pdks will not give you the right result.  I copied the modified script to skywater-pdk-audit and committed it to my branch, so if you do a pull it should work now.

proppy, 2022-07-28:
> The README file states that I needed to make some minor changes to the mismatch_params.py and process_params.py files for the different syntax

I should probably have read it more carefully :)

I made this change in the meantime, that managed to get me to the end of the script:
https://github.com/RTimothyEdwards/open_pdks/commit/014acb7f7afbc34021d4c01bbe8ed1b1196fa72b

proppy, 2022-07-28:
> I copied the modified script to skywater-pdk-audit and committed it to my branch, so if you do a pull it should work now.

Thanks I pulled your copy and got rid of my cargo-cult'ed patch :)

RTimothyEdwards, 2022-07-28:
@proppy :  Okay, thanks.  I do need to update the scripts in the master branch of open_pdks, but this is the quicker/simpler solution for now.

proppy, 2022-07-28:
Ran the tests:
```
Simulation summary:

Passing devices:
sky130_fd_pr__cap_mim_m3_1
sky130_fd_pr__cap_mim_m3_2
sky130_fd_pr__esd_nfet_05v0_nvt
sky130_fd_pr__esd_nfet_g5v0d10v5
sky130_fd_pr__esd_pfet_g5v0d10v5
sky130_fd_pr__nfet_01v8
sky130_fd_pr__nfet_01v8_lvt
sky130_fd_pr__nfet_03v3_nvt
sky130_fd_pr__nfet_05v0_nvt
sky130_fd_pr__nfet_20v0
sky130_fd_pr__nfet_20v0_iso
sky130_fd_pr__nfet_20v0_nvt
sky130_fd_pr__nfet_20v0_zvt
sky130_fd_pr__nfet_g5v0d10v5
sky130_fd_pr__nfet_g5v0d16v0
sky130_fd_pr__pfet_01v8
sky130_fd_pr__pfet_01v8_hvt
sky130_fd_pr__pfet_01v8_lvt
sky130_fd_pr__pfet_20v0
sky130_fd_pr__pfet_g5v0d10v5
sky130_fd_pr__pfet_g5v0d16v0
sky130_fd_pr__diode_pd2nw_05v5
sky130_fd_pr__diode_pd2nw_05v5_esd
sky130_fd_pr__diode_pd2nw_05v5_esd_v5
sky130_fd_pr__diode_pd2nw_05v5_hvt
sky130_fd_pr__diode_pd2nw_05v5_lvt
sky130_fd_pr__diode_pd2nw_11v0
sky130_fd_pr__diode_pw2nd_05v5
sky130_fd_pr__diode_pw2nd_05v5_defet
sky130_fd_pr__diode_pw2nd_05v5_esd
sky130_fd_pr__diode_pw2nd_05v5_esd_v5
sky130_fd_pr__diode_pw2nd_05v5_lvt
sky130_fd_pr__diode_pw2nd_05v5_nvt
sky130_fd_pr__diode_pw2nd_11v0
sky130_fd_pr__model__parasitic__cap_p2ps
sky130_fd_pr__model__parasitic__diode_pd2nw
sky130_fd_pr__model__parasitic__diode_ps2dn
sky130_fd_pr__model__parasitic__diode_ps2dn__extended_drain
sky130_fd_pr__model__parasitic__diode_pw2dn
sky130_fd_pr__model__parasitic__diode_pw2dn_defet
sky130_fd_pr__photodiode
sky130_fd_pr__npn_05v5_W1p00L1p00
sky130_fd_pr__npn_05v5_W1p00L2p00
sky130_fd_pr__npn_11v0_W1p00L1p00
sky130_fd_pr__pnp_05v5_W0p68L0p68
sky130_fd_pr__pnp_05v5_W3p40L3p40
sky130_fd_pr__res_generic_nd
sky130_fd_pr__res_generic_pd
sky130_fd_pr__res_iso_pw
sky130_fd_pr__res_high_po
sky130_fd_pr__res_xhigh_po

Failing devices:
sky130_fd_pr__cap_var_hvt
   Reason: Error: unknown subckt: x1 n3 n4 0 sky130_fd_pr__cap_var_hvt

sky130_fd_pr__cap_var_lvt
   Reason: Error: unknown subckt: x1 n3 n4 0 sky130_fd_pr__cap_var_lvt

sky130_fd_pr__esd_nfet_01v8
   Reason: Error: unknown subckt: x1 3 2 0 0 sky130_fd_pr__esd_nfet_01v8 w=20.35 l=0.165 m=1

sky130_fd_pr__pfet_01v8_mvt
   Reason: Error: unknown subckt: x1 3 2 5 5 sky130_fd_pr__pfet_01v8_mvt w=1.68 l=0.15 m=1

[... 351 lines cut ...]
sky130_fd_pr__cap_vpp_11p5x11p7_m1m2_shieldl1
   Reason: Error: unknown subckt: x1 n3 n4 0 sky130_fd_pr__cap_vpp_11p5x11p7_m1m2_shieldl1

sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3_shieldl1
   Reason: Error: unknown subckt: x1 n3 n4 0 sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3_shieldl1

sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3_shieldl1m5_floatm4
   Reason: Error: unknown subckt: x1 n3 n4 0 0 sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3_shieldl1m5_floatm4

sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3m4_shieldl1m5
   Reason: Error: unknown subckt: x1 n3 n4 0 0 sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3m4_shieldl1m5

sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3m4_shieldm5
   Reason: Error: unknown subckt: x1 n3 n4 0 0 sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3m4_shieldm5

sky130_fd_pr__cap_vpp_11p5x11p7_m1m4_noshield
   Reason: Error: unknown subckt: x1 n3 n4 0 sky130_fd_pr__cap_vpp_11p5x11p7_m1m4_noshield

sky130_fd_pr__cap_vpp_44p7x23p1_pol1m1m2m3m4m5_noshield
   Reason: Error: unknown subckt: x1 n3 n4 0 sky130_fd_pr__cap_vpp_44p7x23p1_pol1m1m2m3m4m5_noshield

Final results:
Passes: 51
Fails:  128
```

proppy, 2022-07-28:
Curious if that's what it's supposed to looks like:

[image]

And if there is way to generate something similar the graph shown @hpretl in https://github.com/google/skywater-pdk/issues/381#issue-1294635084 to compare?

RTimothyEdwards, 2022-07-28:
@proppy :  That all looks correct.  The tests are supposed to fail on the set of devices that are _not_ included in the continuous models, which (as you can see) is quite a lot of them.  All of the "fixed layout" devices (RF devices, parallel plate capacitors, etc.) are among these---there is only one layout, so only one point to characterize, so the existing model in the sky130 PDK is valid.  There were previously 5 fixed-width resistor types for the "high" and "xhigh" poly resistors, so those are no longer valid models, as the continuous model subsumes all of the fixed-width types.  There are no continuous models for the special SRAM devices, nor for the medium-Vt FET (which as far as I know is a valid device, so I don't know why it wasn't included in the continuous models).

FYI, you can pass a filename of devices to test to the `run_spice_tests.py` script.  There is a file in the test/ directory called "continuous.txt" which is a list of all the devices included in the continuous models.  So if you pass that file as an argument to the script, you should get (I think) 51 passes and 0 fails.

The SVG plot is correct.  There is a corresponding plot `sky130_fd_pr__nfet_01v8_lvt_ids_v_vds.svg` that is the set of curves for Ids vs. Vds (instead of Vgs, which is the plot you posted).

Yes, it would be great to replicate @hpretl 's graph with the continuous models.  The easiest way is to get the SPICE netlist that Harald used to generate that graph.

proppy, 2022-07-28:
>  So if you pass that file as an argument to the script, you should get (I think) 51 passes and 0 fails.

Indeed!

```

proppy, 2022-09-01:
I'm back from leave and should be able to make progress on this!

Here is my plan based on the past discussions on this:

1. Reproduce the plot from https://github.com/google/skywater-pdk/issues/381#issuecomment-1220698670 using the continuous model.
2. Verify @hpretl and @bmurmann that they indeed fix the issues pointer in #381.
3. Create a pull request against https://github.com/google/skywater-pdk-libs-sky130_fd_pr that simply add the continuous model next to the binned ones next in [`models/`](https://github.com/google/skywater-pdk-libs-sky130_fd_pr/tree/main/models).
5. Iterate on the pull request with @RTimothyEdwards to organize the models in a way that's friendly for https://github.com/RTimothyEdwards/open_pdks to integrate with.

@RTimothyEdwards @mithro @QuantamHD any issues with that plan?

RTimothyEdwards, 2022-09-01:
@proppy :  None at all.  Sounds like a great plan.

proppy, 2022-09-06:
@RTimothyEdwards please have a look at https://github.com/google/skywater-pdk/issues/381#issuecomment-1237564571, the continuous models don't seems to fix the continuity issue reported in #381. Should we still move forward with the the remaining items (the continuous model do seems to be a lot faster!)

hpretl, 2022-09-06:
@proppy I think it still makes sense to release the continuous models (they have advantages).

We can deal with #381 later by an updated PMOS model (or whichever models show the discontinuous behavior).

proppy, 2022-09-07:
@RTimothyEdwards mentioned in https://github.com/google/skywater-pdk/issues/381#issuecomment-1238339035

>  I recall that the original "bad behavior" demonstration was a plot of something vs. transistor width that showed a discontinuity at the bin boundary

@hpretl @bmurmann do we already have a ngspice test bench that exhibit that? is it possible to sweep across device parameter the same way that you usually swept across voltage?

RTimothyEdwards, 2022-09-07:
@proppy :  Sweeping across a device parameter can be done with an `alter` command and then construct an array from the results of all the loops, but it might be easier just to run multiple short simulations and then concatenate all the results together in a graph using something like octave or matplotlib in python.

bmurmann, 2022-09-07:
@proppy, I don't have a testbench for the width discontinuity problem, but what @RTimothyEdwards suggests should work.

In any case, I am all for releasing the continuous models as soon as possible as mentioned by others above.

proppy, 2022-10-12:
Sorry for the lack of update, as suggested by @RTimothyEdwards in https://github.com/google/skywater-pdk/issues/380#issuecomment-1239331284, I tried to produce a test bench to sweep across bins boundary but couldn't find the right set of parameters to highlight the problem discussed here:

See below a sweep of `w: 1 → 10` with `0.5` increments:
[image]

generated with following notebook: https://colab.research.google.com/gist/proppy/94429878c7c9c60e60f499f7368e17b1/sky130-pyspice-playground.ipynb

@RTimothyEdwards @bmurmann @hpretl could you suggested range of parameters that should highlight the poor behavior of the binned models?

proppy, 2022-10-12:
when varing both `w` and `l`:
[image]

bmurmann, 2022-10-12:
@proppy you may need finer steps in W, L and plot the current versus W or L for some fixed bias condition.

proppy, 2022-10-12:
@bmurmann I think I've got something!
[image]

See https://colab.research.google.com/gist/proppy/94429878c7c9c60e60f499f7368e17b1/sky130-pyspice-playground.ipynb#scrollTo=H1Z5Ro7nPKdp

hpretl, 2022-10-12:
What is the step width of W? Maybe this is too course, or it is indeed model discontinuities. Sometimes model issues are easier seen in the derivatives, e.g. 2nd order taken on Ids-vs-Vgs.

proppy, 2022-10-12:
> What is the step width of W?

It was actually `l: 0.15 → 0.35 +=0.01` in `uM` (the axe was misslabeled).

RTimothyEdwards, 2022-10-12:
@proppy :  The graph X axis should be labeled "L", not "W", since you are iterating over length, not width.
The model bins for the lengths have boundaries at 0.15um, 0.18um, 0.25um, 0.5um, 1um, 2um, 4um, 8um, 20um, and 100um.
There are definitely weirdnesses in the modeling going on at the two boundary points which are captured by your analysis, which is at 0.18um and 0.25um.  Can you analyze out further?

RTimothyEdwards, 2022-10-12:
@proppy :  But the key will be running the same analysis on the nFET continuous model.

bmurmann, 2022-10-12:
@proppy then this curve is probably fine, just showing SCE and reverse SCE. I would repeat with W and fine step.

proppy, 2022-10-12:
> @proppy : The graph X axis should be labeled "L", not "W", since you are iterating over length, not width.

Yep, my bad I miss-labeled the axe! 👹 

> The model bins for the lengths have boundaries at 0.15um, 0.18um, 0.25um, 0.5um, 1um, 2um, 4um, 8um, 20um, and 100um.

Do you also have the bins value for `W`? or even better, is there a way for me to deduce them from the PDK files?

proppy, 2022-10-12:
> just showing SCE and reverse SCE

Should I reduce the bias to minimize it? (it's currently `1.8V`)

proppy, 2022-10-12:
> @proppy : But the key will be running the same analysis on the nFET continuous model.

Yes, that's what I want to do next!

proppy, 2022-10-12:
> Can you analyze out further?

[image]
`W: 10`
`L: [0.15, 0.17, 0.18, 0.19, 0.24, 0.25, 0.26, 0.4, 0.5, 0.6, 0.9, 1, 1.1, 1.9, 2, 2.1, 3.9, 4, 4.1, 7, 8, 9, 15, 20, 25, 90, 100]`

See https://colab.research.google.com/gist/proppy/8f92b08c7db874418cb5cc629f4bcf14/sky130-pyspice-playground.ipynb#scrollTo=0dZxZFhJZFxR

bmurmann, 2022-10-12:
> > just showing SCE and reverse SCE
> 
> Should I reduce the bias to minimize it? (`vdd` is currently maxed at `1.8V` in my testbench)

1.8V should be fine.

proppy, 2022-10-12:
> I would repeat with W and fine step.

@bmurmann do you have an idea of the range? I don't see much going on between `W: 10 → 30`:
[image]

RTimothyEdwards, 2022-10-12:
@proppy :  The nFET 0.18V device is very finely binned and might not be the best model to test.  The boundaries of the width bins are (in microns):
0.36, 0.39, 0.42, 0.52, 0.54, 0.55, 0.58, 0.60, 0.61, 0.64, 0.65, 0.74, 0.84, 1.0, 1.26, 1.68, 2.0, 3.0, 5.0, 7.0, and 100.
So, no, there's not much going on between 10 and 30.

RTimothyEdwards, 2022-10-12:
@proppy:  To find the boundaries in the PDK files, I was looking at `sky130_fd_pr__nfet_01v8__tt.pm3.spice` (any of the corner files will do;  I think they all have the same bins).
Each bin is bounded by `lmin`, `lmax`, `wmin`, and `wmax`.
For example, in the file cited above, line 34 is
```
* Model Flag Parameters
+ lmin = 2.0e-05 lmax = 0.0001 wmin = 7.0e-06 wmax = 0.0001
```
Which means that the binned length is between 20 and 100 microns, and the binned width is between 7 and 100 microns.

proppy, 2022-10-12:
another suggested by @StefanSchippers:
[image]

- `Vg: 0 → 1.8V += 0.01`
- `Vd: 0.1V`
- `L: 0.15 → 1 += 0.15`
- `W: 1`

ericsmi, 2022-10-12:
@proppy  In addition or @hpretl 's suggestion to plot the derivative you might also consider making the y-axis log.   This will make it easier to spot issues in the smaller dimensions.

proppy, 2022-10-12:
> The boundaries of the width bins are (in microns):
0.36, 0.39, 0.42, 0.52, 0.54, 0.55, 0.58, 0.60, 0.61, 0.64, 0.65, 0.74, 0.84, 1.0, 1.26, 1.68, 2.0, 3.0, 5.0, 7.0, and 100.

- `W: 0.36 → 1.0  += 0.1`
- `L: 0.15`
- `Vd: 1.8V`

https://colab.research.google.com/gist/proppy/94429878c7c9c60e60f499f7368e17b1/sky130-pyspice-playground.ipynb#scrollTo=9scIFl1w7Lk6

[image]

RTimothyEdwards, 2022-10-12:
@proppy :  Still not a great test since the bins in this region are tighter than your step size of 0.1um.  I'm still looking for boundary discontinuities specifically;  i.e., if the boundary is at 0.74um then I would like to see the measurements at 0.73um and 0.75um.  The nonmonotonicity around W=0.7um in the graph above does look odd, though.

proppy, 2022-10-12:
@RTimothyEdwards just realized that the plot were using `Vg: 0` (forgot to set it to something after switching from `dc` to `op` simulation) with a fixed `Vd: 1.8`, I guess I should retry with something positive on `Vg`.

proppy, 2022-10-12:
> if the boundary is at 0.74um then I would like to see the measurements at 0.73um and 0.75um

make sense, I was trying to cover a wider area to try to spot a pattern rather than focus on a specific boundaries.

I will craft something manually similar to what I did for `L` in https://github.com/google/skywater-pdk/issues/380#issuecomment-1276594329

proppy, 2022-10-12:
>  you might also consider making the y-axis log

Indeed!
[image]

https://colab.research.google.com/gist/proppy/a0c5ed3e28e942f1621200dcf67bad5a/sky130-pyspice-playground.ipynb#scrollTo=q0XHBAt1jGmQ

proppy, 2022-10-12:
>  I'm still looking for boundary discontinuities specifically

- `W: [0.36, 0.37, 0.38, 0.39, 0.40, 0.41, 0.42, 0.43, 0.51, 0.52, 0.53, 0.54, 0.545, 0.55, 0.56, 0.57, 0.58, 0.59, 0.60]`
- `L:  0.15`
- `Vd: 1.8V`
- `Vg: 1.0V`

[image]

[image]

proppy, 2022-10-13:
@lhrodovalho shared more plot on `open-source-silicon.dev/#analog-design` with a faster testbench (that leverage `alterparam` instead of re-running the full simulation).

[image]
[image]
[image]
[image]

consolidated the testbed and plots in this notebook:
https://colab.research.google.com/gist/proppy/68a62fd4966182fad8f2eb054bddc0e1/sky130-ngspice-playground.ipynb

proppy, 2022-10-21:
One good news and one bad news:
- I checked in the models in https://github.com/google/skywater-pdk-libs-sky130_fd_pr/pull/3
- this does make the simulation faster but doesn't seems to affect the result we found in https://github.com/google/skywater-pdk/issues/380#issuecomment-1277755994

See https://colab.research.google.com/gist/proppy/1b8eb787be44025e75bf6c5a9aa652fd/sky130-ngspice-playground.ipynb

[image]
[image]
[image]
[image]

[4 more replies]