google/skywater-pdk #382: sdf_cond missing in hd libraries

maliberty, 2022-07-11:
## Expected Behavior
sdf_cond statements should be present on conditional arcs in Liberty.

## Actual Behavior
They are present in the hs library but not the hd library.

## Steps to Reproduce the Problem

Compare the [hs library](https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hs/blob/1d051f49bfe4e2fe9108d702a8bc2e9c081005a4/cells/mux2/sky130_fd_sc_hs__mux2_2__tt_025C_1v80_ccsnoise.lib.json) which has sdf_cond for mux2_1 to the [hd library](https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/blob/ac7fb61f06e6470b94e8afdf7c25268f62fbd7b1/cells/mux2/sky130_fd_sc_hd__mux2_2__tt_025C_1v80.lib.json) which does not.

## Specifications

- Version: latest (per link in this repo)
- Platform:  any

proppy, 2022-07-28:
Did a quick survey of all the `sc` libraries:
```
🍡 grep sdf_cond sky130_fd_sc_hd/latest/cells/mux2/*.lib.json | wc -l
0
🍙 grep sdf_cond sky130_fd_sc_hdll/latest/cells/mux2/*.lib.json | wc -l
264
🍊 grep sdf_cond sky130_fd_sc_hs/latest/cells/mux2/*.lib.json | wc -l
378
🍡 grep sdf_cond sky130_fd_sc_hvl/latest/cells/mux2/*.lib.json | wc -l
96
🍙 grep sdf_cond sky130_fd_sc_lp/latest/cells/mux2/*.lib.json | wc -l
672
🍡 grep sdf_cond sky130_fd_sc_ls/latest/cells/mux2/*.lib.json | wc -l
342
🍙 grep sdf_cond sky130_fd_sc_ms/latest/cells/mux2/*.lib.json | wc -l
306
```

Do you know if we expect every `*.lib.json` for each standard cell to have at least one `sdf_cond` directive, or if "it depends" :)

proppy, 2022-07-28:
It also seems that `sdf_cond` is missing for all the cells in `hd`:
```
🍊 grep sdf_cond sky130_fd_sc_hd/latest/cells/*/*.lib.json | wc -l
0
```